olikraus/u8g2 #208: SSD1329@128x96-pixels - not more than 64-pixels possible

creesoo, 2017-03-05:
Hey friends,

I'm working on the U8G2lib to support my SSD1329U2-based OLED-Display. What I've done:

1.) Abused the u8x8_d_ssd1325_128x64_nhd_init_seq with a modified display-offset (from 0x04c to 0x000)
2.) Changed the display-height at the end of u8x8_dssd1325.c from 64 to 96 pixels. 

BUT: Still, it's not possible, to display more then 64 lines on the OLED-display. No matter, if I want to show a string or a rectangle - no output > line 64. If I return to the default display-offset-setting-value of 0x04c the graphic content gets cut out UPSIDE of line 64, but appears perfectly down to line 96.
What could be wrong? Any help would be appreciated!

Best, 
Chris

olikraus, 2017-03-06:
Can you attach details of the OLED?
I will create a new display for this: There are some more places to change the width.

creesoo, 2017-03-06:
Hi Oli, 

Thanks for yor quick reply. It is a RiTdisplay RGC13128096WH with the SSD1329U2-Controller, controlled via SPI! As this special-display is a surplus-part, there is no information available except my reverse-engineered pinout (with the usual dataconnections: cs, rs, d/c, sclk, sdin). But the controller is also used by manufacturers like Bolymin. The datasheet can be found here: http://www.displayfuture.com/Display/datasheet/controller/SSD1329.pdf 

Regards, 
Chris

olikraus, 2017-03-06:
So from your research so far, you suggest to reuse the ssd1325 code?
If yes, then i will create a new file for the SSD1329 based on the SSD1325. We can then modify and update the copy.

creesoo, 2017-03-06:
Yes, the SSD1325 seems to be compatible to the 1329, except the problems, I mentioned above!

olikraus, 2017-03-06:
Hmm... I am in the porting process. Commands still differ a little bit. Especially the graphics acceleration command 0x024 is not documented in the SSD1329 datasheet. Can you run some of the 8x8 examples to check whether this works?

olikraus, 2017-03-06:
ok, you can download the lib here: 
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Install this via the Arduino IDE (add zip lib)

The SSD1329 constructor is also part of the examples.
Can you run the u8x8/FlipMode example? It changes the orientation periodically. It also will show, whether the acceleration command works.

creesoo, 2017-03-06:
Thank you very much for your efforts! I will run the examples tomorrow and give you a feedback!

creesoo, 2017-03-08:
Hi Oli! I've finally tested the library in combination with the SSD1329. All 8x8-examples (except the Arduboy, that can't be used, as i am working on a bare Pro-Mini) work besides the fact that the library still only displays line 65-96. The upper part of the OLED (1-64) still remains dark.... Best, Chris

olikraus, 2017-03-08:
ok, i missed to change the 0x04c offset of the 0x0a2 command. Can you retest the new version?

https://github.com/olikraus/U8g2_Arduino/archive/master.zip

creesoo, 2017-03-09:
Now the OLED shows line #1-64. But I noticed one special detail, regarding the MessageBox-demo: In the former version of the lib, the graphic-output was cut at Line 128. Now, everything gets displayed, alligned at line 64.

olikraus, 2017-03-09:
Can you attach a picture how it looks at the moment? Maybe a picture of the "u8g2 logo"?

creesoo, 2017-03-09:
Pictures of several pagebuffer-demos, including the outout of the HelloWorld-script, which I rewrote to display "Hello world!" for 9 times....
[image]
[image]
[image]
[image]

olikraus, 2017-03-10:
I have checked the code, but i have no idea what could restrict the height ot 64.

olikraus, 2017-03-10:
I have created another test release, which shifts the content down by 32 pixel. My hope would be to see some text below the 64 bit line.

creesoo, 2017-03-10:
That didn't work as well - no output beyond after line 64, much rubbish from pixel 0~32. The good news: I solved the problem after rereading the datasheet. The "multiplex ratio" A8h implies the wanted number of vertical pixels (in this case 0~95). So, U8X8_CA(0x0a8, 0x05f) does the job in combination with a zero display-offset.
[image]
The working u8x8_d_ssd1329.c: 

[u8x8_d_ssd1329.zip](https://github.com/olikraus/u8g2/files/834497/u8x8_d_ssd1329.zip)

olikraus, 2017-03-10:
wow, great finding
I have updated the master.zip file

closed 2017-03-11 by olikraus