olikraus/u8g2 #209: LCD Module support

schmurfy, 2017-03-06:
This is not really an issue but I don't know where to ask, I bought one of these modules: http://www.icstation.com/gdc0570-dynamic-driving-type-display-401428mm-screen-p-9672.html
I am still not clear how to drive these and ultimately would love to use this library to drive it, I initially thought they were driven like basic 7 segment displays but that's not the case, if I apply 3v to two random pins I can light a segment but it fades out after a few seconds, I can then make it light again later...

Does anyone know how it is driven and if I can interface it with this library ? Or alternatively if you know a supported driver I can use with it I also take it, this seems like a really nice screen.

olikraus, 2017-03-06:
No, this is a library for graphics LCD and OLEDs like this:
http://www.icstation.com/mobile/13in-blue-oled-display-module-interface-128x64-arduino-p-10988.html

schmurfy, 2017-03-07:
my bad, I got confused xD

closed 2017-03-07 by schmurfy