olikraus/u8g2 #210: Particular font size support

Toor0Root, 2017-03-06:
Hello,

I plan to buy a 240x128 LCD and I inquire on the different libraries. I think your library is the best.

I would like to do my own font where each character is 16x8 (like an alphanumeric LCD), and I want to know, is it possible (and if yes, how) ?

Best regards

olikraus, 2017-03-07:
There is a tool linux bdfconv, which reads a bdf file and converts this to the internal format. Have a look in the issues here, it had been discussed several times.

Toor0Root, 2017-03-07:
Thank you for your answer.

I saw the other issues but I was wondering if each character have a definided width (that can't be changed), or the width adjusts automatically to fit the width of the character itself (a "I" takes less place in width than a "A") ?

olikraus, 2017-03-07:
The width is defined for each glyph in the bdf input file.

closed 2017-03-14 by Toor0Root