olikraus/u8g2 #211: uc1604 jxl19264 doesn't work on nodemcu

disassembly, 2017-03-09:
Hi, I found that an uc1604 jxl19264 lcd can be driven on arduino uno, however, on an nodemcu, with the same code and ide, it doesn't work. What could be the problem? By the way, an pcd8544 lcd works on both boards.

olikraus, 2017-03-09:
hmm... did you check the signals with a scope?
Actually i have no idea other than a wiring problem.

disassembly, 2017-03-09:
I will check it carefully and  report later. Thank you.

nurfaizfoat, 2017-03-10:
@disassembly can you share your wiring hookup here? I managed to get ST7920 on an ESP2866 based board (Wemos D1)

disassembly, 2017-03-10:
Yes, of course. Here is the wiring: (left nodemcu, right uc1604 19264 lcd)
D1 (GPIO5)<---> SCLK
D2 (GPIO4) <---> Data
D5 (GPIO14)<---> DC
D6 (GPIO12)<---> CS
D7 (GPIO13)<---> RESET
constructor is :
U8G2_UC1604_JLX19264_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ D_1, /* data=*/ D_2, /* cs=*/ D_6,  /* dc=*/ D_5,/* reset=*/ D_7);
with a definition:
#define D_0 16
#define D_1 5
#define D_2 4
#define D_3 0
#define D_4 2
#define D_5 14
#define D_6 12
#define D_7 13
#define D_8 5
#define D_9 3
#define D_10 1

disassembly, 2017-03-10:
I finally make it work accidentally. I think the problem is the circuit, because though it works, the display is unstable. Maybe due to the power or something else.
The lcd is not bought commercially but  a part of some other device. So the lcd pins are directly soldered and connected to the nodemcu pins without any circuit change board. I have to charge the device with the origin power supply and the power on the nodemcu to get the lcd display.
Anyway, it is for fun！

olikraus, 2017-03-10:
ok, so it can be closed?

disassembly, 2017-03-10:
Yeah!

closed 2017-03-10 by disassembly