olikraus/u8g2 #212: SSD1606 drawXBM

What7777, 2017-03-10:
Drawing a Bitmap looks completly wrong.
I am not sure, but it completly wrong an partitial mirrored. I can't realized where the error is. Is it because of the BLV format Assembly Electronics use?

PS: 
https://github.com/olikraus/u8g2/wiki/u8g2reference#drawxbm

There are two small errors in Wiki:

[...] u8g.drawXBM( 0, 0, u8g_logo_width, u8g_logo_height, u8g_logo_bits);

    Arguments:
        u8g : Pointer to the u8g structure (C interface only).
        x: X-position.
[...]
u8g is wrong -> it should be u8g2

JamesGKent, 2017-03-10:
you are correct that there is a copy/paste error on the wiki, but as for your issue, try the drawBitmap function:
[https://github.com/olikraus/u8g2/wiki/u8g2reference#drawbitmap](https://github.com/olikraus/u8g2/wiki/u8g2reference#drawbitmap)

JamesGKent, 2017-03-10:
see also a previous issue:
[Issue 188](https://github.com/olikraus/u8g2/issues/188)

What7777, 2017-03-10:
a 32x32 xbm where only the border 1px is black:
u8g2.drawBitmap(32, 32, 32, 32, test_pf);   --> draw nothing
u8g2.drawXBM( 32, 32, 32, 32,test_pf);      --> draw something randomly (two different that still look always ) and even if change the bitmap it still draw exactly the same random thing

JamesGKent, 2017-03-10:
i notice that you are storing your bitmap in progmem, have you tried:  
`u8g2.drawXBMP( 64, 32, 32, 32,test_pf);`  
from the wiki:  

> The XBMP version of this procedure expects the bitmap to be in PROGMEM area (AVR only).

and also:

> Note: The XBMP version requires, that the bitmap array is defined in this way:
static const unsigned char u8g_logo_bits[] U8X8_PROGMEM = { ...

What7777, 2017-03-10:
Same issue! The weirdest thing is that the top 0x0-->31x15 changing constantly with the begin of the loop, while 16x0-->31x31 is constant (stable)!

JamesGKent, 2017-03-10:
strange, so is this code (with your constructor) what you've just tested?
```
void loop(void) {
  u8g2.firstPage();
  do {
    //u8g2.drawBitmap(32, 32, 32, 32, test_pf);
    u8g2.drawXBMP( 32, 32, 32, 32,test_pf);
    
  } while ( u8g2.nextPage() );
  delay(4000);
}

#define test_pf_width 32
#define test_pf_height 32
static const unsigned char test_py[] U8X8_PROGMEM = {
   0xff, 0xff, 0xff, 0xff, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0xff, 0xff, 0xff, 0xff };
```

What7777, 2017-03-10:
I only wanted to test the function of the bitmap, because I want later draw Icons on the display.

olikraus, 2017-03-10:
Thanks for pointing out the copy/paste error. I fixed the wiki meanwhile.

@What7777: Maybe you can post a complete example. To me it is not really clear what is wrong here.

What7777, 2017-03-11:
```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
U8G2_SSD1606_172X72_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);		// eInk/ePaper Display

#define test_pf_width 32
#define test_pf_height 32
static const unsigned char test_py[] U8X8_PROGMEM = {
   0xff, 0xff, 0xff, 0xff, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 0x80,
   0x01, 0x00, 0x00, 0x80, 0xff, 0xff, 0xff, 0xff };

void setup(void) {
  u8g2.begin();
  u8g2.setFlipMode(0);
}

void loop(void) {
  u8g2.firstPage();
  do {
    //u8g2.drawBitmap(32, 32, 32, 32, test_pf);  //  draw nothing!
    u8g2.drawXBMP( 32, 32, 32, 32,test_pf);
    
  } while ( u8g2.nextPage() );
  delay(4000);
}
```
I want test the function, because i want use it for later to draw for example a WiFi-icon
I expect a square where only the edge is black.

And I get this result: 
http://i.imgur.com/mBSx7m6.png 
Where all above the horizontal line is random at every 4sec!

olikraus, 2017-03-11:
maybe your display is incompatible. i mean, the chip might be the ssd1606, but maybe init code has to be different. Can you post the datasheet for your display?

What7777, 2017-03-11:
Here:
http://www.lcd-module.com/products/epaper.html
http://www.lcd-module.com/fileadmin/eng/pdf/grafik/epa20-ae.pdf
All other things like the graphics test work correctly or the logo-demo.

olikraus, 2017-03-11:
oh....
ok, i guess, i have to do some tests with my own epaper device

olikraus, 2017-03-11:
I have just tested the XBM example with both DrawXBM and DrawXBMP.
Everything works as expected.
There is no problem or bug.

What7777, 2017-03-11:
Can you give your source code? I will test it on Monday!

olikraus, 2017-03-12:
It is part of the U8g2 examples in the Arduino IDE.

olikraus, 2017-03-12:
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/XBM/XBM.ino

What7777, 2017-03-13:
I updated to 2.14.7 and everthing works fine now! Make a rollback on 2.13.5 and it doesn't work!

By the way the 2.14.7 in the ArduinoIDE doesn't contain all u8g2_page_buffer examples (https://github.com/olikraus/u8g2/tree/master/sys/arduino/u8g2_page_buffer): Clock, PowerSaveTest, XBM and XORTest

What7777, 2017-03-13:
a

closed 2017-03-13 by What7777

olikraus, 2017-03-13:
Thanks for the hint regarding the examples. I will crosscheck this.