olikraus/u8g2 #213: Support for another SSD1306 SPI

madneon, 2017-03-11:
Is it possible to support such OLED with those 7 connectors?
1. GND
2. VCC 3.3-5V
3. Data0
4. Data1
5. Reset (low active)
6. Data/Command (low for command, high for data)
7. ChipSelection (low active)

olikraus, 2017-03-11:
This is a classic 4-wire SPI OLED. It is already configured as 4-Wire SPI.
D0 is the clock line and D1 is the data (MOSI) line.

madneon, 2017-03-12:
Ah, thanks for clarification!

In that case all of below constructors are valid?
U8G2_SSD1306_128X64_NONAME_1_4W_SW_SPI(rotation, clock, data, cs, dc [, reset]
U8G2_SSD1306_128X64_NONAME_2_4W_SW_SPI(rotation, clock, data, cs, dc [, reset])
U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI(rotation, clock, data, cs, dc [, reset])
U8G2_SSD1306_128X64_VCOMH0_1_4W_SW_SPI(rotation, clock, data, cs, dc [, reset])
U8G2_SSD1306_128X64_VCOMH0_2_4W_SW_SPI(rotation, clock, data, cs, dc [, reset])
U8G2_SSD1306_128X64_VCOMH0_F_4W_SW_SPI(rotation, clock, data, cs, dc [, reset])

olikraus, 2017-03-12:
Additionally, also the hardware constructors are vailid:
U8G2_SSD1306_128X64_NONAME_1_4W_HW_SPI u8g2(rotation,cs, dc [, reset]);
...

SW vs. HW:
HW constructor will use the SPI subsystem of your uC, which is usually much faster. Pins are predefined by the hardware and not required as arguments
SW constructor will emulate SPI protocol within software.

1 vs. 2 vs. F
F is fastest, but will require most RAM. Additionally you do not require the FirstPage/NextPage loop
1 is slower, but will require much lesser RAM

NONAME vs. VCOMH0
Different init sequence, you have to figure out what will work with your display.
NONAME should work in general, VCOMH0 maximizes the range which can be set by the setContrast command.

madneon, 2017-03-12:
My OLED works now with VCOMH0 constructor in Framebuffer HW mode :)

`U8G2_SSD1306_128X64_VCOMH0_F_4W_HW_SPI u8g2(U8G2_R0, 10, 9, 8);`

Can you please clarify how exactly pins are predefined and not required as arguments? My OLED works only if pins 13 and 11 are connected (as it was working in SW mode). Is it normal?

olikraus, 2017-03-12:
Depends you your Board. The Arduino Uno has a atmega328 controller, which has the HW SPI subsystem output at pins 13 and 11. This means, to answer this question you have to consult the uC manual/datasheet. 

For some official boards, this is also listed here:
https://www.arduino.cc/en/Reference/SPI

madneon, 2017-03-12:
I understand now, thank you very much!

closed 2017-03-13 by olikraus