olikraus/u8g2 #214: git version tags

tcurdt, 2017-03-12:
The log suggests there are versions (2.4.17) for the lib - but the repo has no releases nor tags.
Did you maybe forgot a `git push --all`?

olikraus, 2017-03-12:
The releases are maintained in a different repo:
https://github.com/olikraus/U8g2_Arduino/releases

Any lib which should appear in the Arduino IDE has to have a specific structure. However this structure differes from the u8g2 project. Thats why there are two github projects: One which is used for development and another project for Arduino Lib Releases.

tcurdt, 2017-03-12:
It still would be nice to have the release tags on the dev repo.

olikraus, 2017-03-12:
True. I try to them in the future.

closed 2017-03-26 by olikraus

RechieKho, 2024-12-11:
True, it still would be nice to have the release tags on the dev repo.

RechieKho, 2024-12-11:
Woah, so cool that I got such quick response!

Unfortunately, I only get the [tag from 3 years ago](https://github.com/olikraus/u8g2/releases/tag/2.29.11). It is not mandatory, but it would still be nice to have a tag so it is easier to include this as a CMake project via FetchContent (at least for me). Did you push the tags (via `git push --tags`) though?

olikraus, 2024-12-11:
Hmm... good point.
Maybe I should do something like: git config --global push.followTags true
in the repo