olikraus/u8g2 #215: Release handbrake on SSD1306 with HW-SPI?

Fuchks, 2017-03-12:
Is there any drawback on setting sck_clock_hz to 10.000.000 in u8x8_display_info_t (u8x8_d_ssd1306_128x64_noname.c line 235)?
The parameter sck_pulse_width_ns is 50ns, so this is set already to 10MHz.
I don't know if this could be adopted for other displays according to their datasheets too...

FPS-Results:
```
SPI @4MHz:
U8G2_SSD1306_128X64_NONAME_1_4W_HW_SPI		Uno		Clip=35.7 Box=77.2  @=5.1 Pix=8.9
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI		Uno		Clip=70.7 Box=96.5  @=11.7 Pix=15.9
SPI @8Mhz (Uno max 8MHz):
U8G2_SSD1306_128X64_NONAME_1_4W_HW_SPI		Uno		Clip=37.2 Box=84.1  @=5.2 Pix=9.0
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI		Uno		Clip=76.5 Box=107.6  @=11.8 Pix=16.2
```

olikraus, 2017-03-13:
hmm... has to be tested...

olikraus, 2017-05-14:
ok, increased to 8MHz, actually it is indeed a bug which as introduced while switching to the 1.6.0 SPI interface.

olikraus, 2017-05-14:
Maybe other display types are also affected, but at least the ST7565 only alows 4MHz max. soo... closing this... maybe check for other SSDs once...

closed 2017-05-14 by olikraus