olikraus/u8g2 #218: st7920 in SPI hw mode stm32

sass20, 2017-03-14:
i have problem with function "Communication Callback "
in this function come row data for st7920 if it work in spi  mode. In Spi mode for sr7290 need 3 byte packet but in function come only one byte.

 I used source from https://github.com/olikraus/u8g2/tree/master/csrc
And second
I compile u8g2_d_memory.c but it used very very many memory ...  All function in this file reserve memory in cpu but it is not used...

olikraus, 2017-03-14:
your byte com procedure will be called by the procedure in line 314 in this file:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_cad.c

cad = command arg data

this level will take care on the protokol issues of the ST7920, you just need to send the byte.

u8g2_d_memory.c: Yes, u8g2 very much depends on the linker garbage collection.
For gcc this includes the use of -ffunction-sections -fdata-sections and later the use of --gc-sections at link time. u8g2_d_memory.c is only one place where this is used. It is also used for the fonts.

sass20, 2017-03-14:
thank you for answer
I found mistake in my LCD setup code and fix it...
About using stm32
I send bytes in dma mode via SPI It work correctly if i send 3 byte only ... (For LCD need time for move recived data  to RAM)
I send bytes in normal mode (no dma no interrupt) and it is work fine I send 30 bytes ))) delay between bytes 62 uS (I measure in logic analizator)

olikraus, 2017-03-26:
so i can close this issue?