olikraus/u8g2 #219: Newhaven NHD‐C12864A1Z‐FSW‐FBW‐HTT

craigroutledge, 2017-03-14:
Apologies if this isn't the right place to submit this, but I can't seem
to find a discussion or help list/group anywhere ....

I've been trying to use a Newhaven LCD (part NHD‐C12864A1Z‐FSW‐FBW‐HTT)
with U8G2 on an Arduino.  It works fine with U8glib, but when I upload U8G2
based code the screen just stays blank and nothing happens.

I've attached the "Hello World" code I used below, which is lifted from the
documentation on the two wikis.

Obviously, I'm not altering (or even touching) the hardware between the
software versions.

The hardware is an Arduino Uno clone (an Itead Iteaduino UNO so that I can
run the board at 3.3V for the LCD module and the project I have in mind).

And yes, it isn't wired to the hardware SPI pins because I'll have to use
software SPI in the eventual project anyways -- so I'm testing that.

Tested with Windows 10, Arduino IDE 1.8.1, U8G2 2.14.7, and U8glib 1.19.1.

First tested under Linux with Fedora 25, Arduino IDE 1.6.4, U8G2 2.14.7, and
U8glib 1.19.1.  This yields the same result.

Any idea what's wrong?   Anything else I can do to help diagnose this?
Thanks for any help.

[u8g2_nhd_code.txt](https://github.com/olikraus/u8g2/files/842055/u8g2_nhd_code.txt)

olikraus, 2017-03-14:
Strange, we just introduced this device. Can you play around with different contrast setting s? 150 should be the correct value.

Unfortunately I do not own this display.

craigroutledge, 2017-03-14:
Thanks, that did it.  A contrast of 150 is too low on my LCD, but 200 is good.

I don't know how much variation there is between individual boards, but for mine I got the following results (if it's of any use to you).

At 150, it is just at the threshold of vision if you squint real hard and know exactly what to look for under bright daylight.  You might convince yourself it's just your imagination.

At 175 it is visible but faint and could be overlooked.

200 looks good to me.

At 225 the background becomes faintly visible but easy to miss.
At 250 the background grid is moderately noticable.

olikraus, 2017-03-14:
I think 150 was good for the previous user requesting this device. But maybe i should increase the default to 175.

olikraus, 2017-05-14:
set default contrast for this display to 180, closing...

closed 2017-05-14 by olikraus