olikraus/u8g2 #221: looking for fonts with polish language characters

XnIcRaM, 2017-03-19:
I can't find font with characters used in my language, I'm looking for ąĄćĆęĘłŁńŃóÓśŚżŻźŹ in UTF-8 it should be at: 
ą - c485
Ą - c484
ć - c487
Ć - c486
ę - c499
Ę - c498
ł - c582
Ł - c581
ń - c584
Ń - c583
ś - c59b
Ś - c59a
ź - c5ba
Ź - c5b9
ż - c5bc
Ż - c5bb
Ó - c393
ó - c3b3

olikraus, 2017-03-26:
I think these are NOT the unicode for, but instead the UTF-8 encoding

ą - 105
Ą - 104
ć - 107
Ć - 106
ę - 119
Ę - 118
ł - 142
Ł - 141
ń - 144
Ń - 143
ś - 15b
Ś - 15a
ź - 17a
Ź - 179
ż - 17c
Ż - 17b
Ó - 0d3
ó - 0d3

olikraus, 2017-03-26:
I will add this for unifont.
Any other fonts?

```
/*
  Fontname: -gnu-Unifont-Medium-R-Normal-Sans-16-160-75-75-c-80-iso10646-1
  Copyright: Copyright (C) 2014 Roman Czyborra, Paul Hardy, Qianqian Fang, Andrew Miller, et al. Licensed under the GNU General Public License; either version 2, or (at your option) a later version, with the GNU Font Embedding Exception.
  Glyphs: 114/57086
  BBX Build Mode: 0
*/
const uint8_t unifont_polish[1555] U8G2_FONT_SECTION("unifont_polish") = 
  "r\0\3\2\5\5\4\5\6\20\20\0\376\12\376\13\377\1\227\3\60\4\365 \6\0\240G\1!\10A"
  "\61DqH\4\42\10\205(F\221\271\5#\17F%D\325\323\60$Q\313\60D=\1$\22G%"
  "D\27\16J\24I\341<FRe\20\63\0%\24G%D\243II)\211\224\64N\23)\211\222\222"
  "\246\0&\22G%D\265\225\262J(&\221\226\210I&M\1'\7\201\60F\61\4(\14\203\355C"
  "\225DI\324[\224\5)\15\203\351C\221EY\324K\224D\0*\15\347dDW\252\264mIS-"
  "\3+\14\347dD\27\327\206!\213k\0,\11\202\254C\241$\12\0-\7$(E\61\4.\7B"
  ",D\61\4/\14F%D[\254\206iXM\1\60\16F%D\245EI\350c\22e\22\0\61\13"
  "E)D\225II\330\247A\62\17F%D\63$\241\230fZXM\207\1\63\20F%D\63$\241"
  "\230Fs*\212\311\220\0\64\20F%D\31jIT\311\222,\31\306\264\2\65\17F%DqH\253"
  "\203\234\246b\62$\0\66\17F%D\65\205i:(\241c\62$\0\67\13F%Dq-\246\305\264"
  "\11\70\20F%D\63$\241\61\31\222\320\61\31\22\0\71\16F%D\63$\241\61\31\324\306h\2:"
  "\11\342lD\61\304C\0;\12\42\355C\61\304J\242\0<\11%)D\231u\355\0=\11\246\244D"
  "q'\16\3>\11%%D\221v\353\10?\17F%D\63$\241\230\206\325\34L#\0@\22F%"
  "D\65eR\242$K\244DJ$-\361\20A\16F%D\245E-\241\70\14\242c\0B\16F%"
  "D\61(\241qXB\307a\1C\16F%D\63$\241\265\243\230\14\11\0D\16F%D\61DY"
  "\22\372-\31\42\0E\15F%DqH\253\203\222\266\16\3F\14F%DqH\253\203\222v\5G"
  "\16F%D\63$\241\265\64\204\66e\11H\13F%D\21:\16\203\350\61I\13E)D\61Ha"
  "?\15\2J\16G%D\65\210qOY\224e\33\0K\21F%D\21jIT\311D\61\311\242Z"
  "\22\6L\11F%D\221\366\327aM\15F%D\21\212\323\20-\36\35\3N\20F%D\21n\233"
  "\22)\221\224H\211v\14O\14F%D\63$\241?&C\2P\15F%D\61(\241qX\322\256"
  "\0Q\27g\345C\63Da\22&a\22&a\22&\211\222H\322\220\3\2R\20F%D\61(\241"
  "qX\242Z\222%\241\30S\15F%D\63$\241\331QL\206\4T\12G%Dq\310\342\376\6U"
  "\13F%D\21\372\307dH\0V\21G%D\221Z\223,\312\242\254\22&i\234\1W\15F%D"
  "\21zq\231\206h\24\3X\17F%D\21\212I\324&jQK(\6Y\16G%D\221\252I\26"
  "e\225\64\356\6Z\13F%Dq-\366\232\16\3[\12\203\361C\61D\375\323\0\134\14F%D\221"
  "\306\325\70\215\253\1]\12\203\345C\61\365OC\0^\11fdF\245EI\30_\7'\344Cq\10"
  "`\7c\250F\221\25a\16\6%D\63$a\232\14\243MY\2b\16f%D\221\266,\232\350\270"
  ")\13\0c\15\6%D\63$\241\332\61\31\22\0d\14f%D\333\262h\243\67e\11e\17\6%"
  "D\63$\241\70\14j\61\31\22\0f\14e%D'\205\245A\12{\2g\23f\245C\233,Z\222"
  "%Y\264\245C\22\212\311\220\0h\14f%D\221\266,\232\350c\0i\13e)D\25\346\210\330\247"
  "Aj\14\245\245CYG\304>J\221\4k\21f%D\221\266%Q%\23\223,\252%a\0l\12"
  "e)D#\366O\203\0m\22\7%D\261(Q$ER$ER$ER\1n\13\6%D\221"
  ",\232\350c\0o\14\6%D\63$\241\217\311\220\0p\16F\245C\221,\232\350\270)K\232\2q"
  "\14F\245C\263h\243\67eI\13r\13\6%D\221,\232\250v\5s\15\6%D\63$\241\354\230"
  "\14\11\0t\13E%D\25\226\6)\354*u\12\6%D\21\372MY\2v\14\6%D\21\32\223"
  "\250\67Q\2w\21\7%D\221J\221\24I\221\24I\221T\261\0x\17\6%D\21\212I\224\211Z"
  "\224\204b\0y\16F\245C\21zL\42KZ\31\22\0z\12\6%Dq\15{\35\6{\16\203\355"
  "C\63eQ\22eQ\22e\2|\7\301\261C\361A}\17\203\351C\241EI\224EI\224E\13\0"
  "~\12g$F\243I\221\246\0#\20\242\203\221\364;I\347\244\223\16I\66%a\32%C\222M"
  "I\230NC\62\354\234tN:)\351\7\323\17\306%D'\351\244!\11\375\61\31\22\0\363\16\206%"
  "D'\351\244!\11}L\206\4\0\0\1\4\26\207\245C%FYTL\302dX\302$L\302$L"
  "s@\1\5\24G\245C\63Da\34\15J\230\204I&-i\16\10\1\6\21\306%D'\351\244!"
  "\11\255\35\305dH\0\1\7\20\206%D'\351\244!\11\325\216\311\220\0\1\30\17\206\245CqH\253"
  "\203\222\266\16[,\1\31\21F\245C\63$\241\70\14j\61\31\262X\1\1A\16G!D\23\227\262"
  "$\64\67\17\3\1B\16e)D#\266D\246$l\32\4\1C\24\306%D'\351\224p\333\224H"
  "\211\244DJ\264c\0\1D\16\206%D'\351\224d\321D\37\3\1Z\21\306%D'\351\244!\11"
  "\315\216b\62$\0\1[\21\206%D'\351\244!\11eK(&C\2\1y\17\306%D'\351\224"
  "a-\366\232\16\3\1z\16\206%D'\351\224a\15{\35\6\1{\17\306%D\225\346\224a-\366"
  "\232\16\3\1|\16\206%D\225\346\224a\15{\35\6\0";
```

olikraus, 2017-03-26:
font rebuild required

closed 2017-05-21 by olikraus

SteveSzyd, 2018-04-07:
Hi
I am trying to use this font u8g2_font_unifont_t_polish, but polish letters are not displayed on LCD (0,96" 128x64 I2C). For example word "Światło" written by u8g2.print command looks like "wiato".  What am I doing wrong? Please help.
Steve

olikraus, 2018-04-07:
You can check the font here: https://github.com/olikraus/u8g2/wiki/fntgrpunifont

Did you enable utf8 support for the print command?

SteveSzyd, 2018-04-09:
> You can check the font here: https://github.com/olikraus/
> u8g2/wiki/fntgrpunifont
>
> Did you enable utf8 support for the print command?

"Did you enable utf8 support for the print command?" !!!!!!
Thank you very much for an answer. Answer for your question is "NO, I
DIDN''T".
I hope command "u8g2.enableUTF8Print();" will solve my problem, but I'll
check it later.
Anyway one more biiiig thanks :)
Steve

SteveSzyd, 2018-04-09:
I have checked idea from my previous post and all is OK. Problem solved.
One more THANK YOU.
Steve

2018-04-09 11:33 GMT+02:00 Stefan Sz <stefanszy@gmail.com>:

olikraus, 2018-04-09:
:)