olikraus/u8g2 #222: Add support for LC7981 with 160x160

Ancalagon94, 2017-03-21:
Hi,
I would like to use a Winstar WG 160x160 display with a LC7981 interface. However, I can't seem to find any library that supports this, the closest being for 160x80 display. If there isn't a simple solution I have overlooked, would it be possible to add support for this resolution?

olikraus, 2017-03-23:
Can be done if you you takeover testing. But I do not have much internet access at the moment. Maybe next week.

olikraus, 2017-03-26:
I have a test version here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Please install this zip file via the "add zip lib" feature of the Arduino IDE. 
Can you sent me some test results of the new driver?

stevie72, 2017-03-27:
Hello olikraus!
I´m using a PCD8544 (Nokia5110). Is there a clearscreen funtion in you Library u8g2 for arduino?
Thanks, Stefan

olikraus, 2017-03-28:
Clear screen: yes, see the reference.

closed 2017-04-09 by olikraus