olikraus/u8g2 #223: Fast I2C in possible combination with an SSD1306 without a reset?

MarsLoop, 2017-03-24:
I'm using the following constructor, but its pretty slow:

U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=/ SCL, / data=/ SDA, / reset=*/ U8X8_PIN_NONE); // OLEDs without Reset of the Display

I know this display has once worked faster with your old library which included a fast i2c option, at least in some examples. 
Could you please add a constructor or tell me how else to enable it?

olikraus, 2017-03-26:
Correct, HW i2c would be much faster

olikraus, 2017-03-26:
Can we close this issue?

MarsLoop, 2017-03-30:
If I just change the "sw" to "hw" it doesn't display anything anymore.
Platform: Arduino Diecimila

My current constructor(?) is:
`U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display`

May the problem is related with a thing I found strange already with the u8x8-samplecode: 
Why don't I have to specify the I2C-adress like in some other samples? I'm a bit irritated right now I have spent 15min search but I couldnt find any example where the example code where the adress was required in the sketchbook or elsewhere but I'm sure there was.

MarsLoop, 2017-03-30:
Sorry I forgot I've switched to U8G2 in this programm already again.  

`U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display`

Doesn't yield any display on the screen.

olikraus, 2017-03-30:
There is a separate function to set the i2c address. See the reference manual for this.

When switching to HW i2c or HW SPI, you must remove clock and data pins from the constructor ARGS. See the setup guide.

closed 2017-04-09 by olikraus

MarsLoop, 2017-04-14:
I haven't had a display anymore to test it, no I tried it but the following constructors don't work:

// Don't work - black screen
U8G2_SSD1306_128X64_NONAME_1_2ND_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); 
U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

//Works but slow
U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);

olikraus, 2017-04-15:
The U8G2_SSD1306_128X64_NONAME_1_HW_I2C constructor should work.

MarsLoop, 2017-04-16:
Do you mean the display is at fault?
That constructor doesn't work: U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, / reset=*/ U8X8_PIN_NONE);

olikraus, 2017-04-16:
But it should. Do you have proper pull up resistors attached to both pins?

MarsLoop, 2017-04-28:
Yes, I've tried with 22k, 3,9k pull-ups. It is a very popular display so many people should be affected. But I'm not sure how many actually try to use hardware i2c, because there is no example constructor in the example files. As far as I know.

combs, 2017-06-26:
@MarsLoop It's there
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_full_buffer/HelloWorld/HelloWorld.ino#L62

Hanser, 2018-10-28:
Using the full buffer version (with an F) solved it for me.

U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);