olikraus/u8g2 #225: 2 Pixel horrizontal offset?

maik-dmxc, 2017-03-25:
I have a problem with the u8g2 and a SSD1306_128X64 display.
on the left side, 2 pixels are lost?
on the rigth side, a strange 2 pixel wide vertikal line appears

the sketch is the full_buffer fps.ino

selected hardware is
`U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
`

same thing with spi display.
same behavior with all other sample sketches.
[image]
[image]

combs, 2017-03-25:
Try the SH1106 initializer instead of SSD1306... I think the SH1106 is very similar but has 132x64 pixels in its internal memory instead of 128x64.

maik-dmxc, 2017-03-26:
solved!!!

Hi comps,

with 
`U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ SCL, /* data=*/ SDA);`
it works for me.

regards Maik

closed 2017-03-26 by maik-dmxc