olikraus/u8g2 #226: ks0108 contructor

sonic2wb, 2017-03-25:
U8G2_KS0108_128X64_1(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, dc, cs0, cs1, cs2 [, reset])

can you tell me what dc stands for. my board has cs1 and cs2 im assumeing i yes cs0 and cs1 and leave cs2 as none. also what pin does R/W need to be on. ur just tie it low? and the D/I pin. what or where is that defined??

i cant get the pcf i2c board i made to work so i thought i would try this first. but still no luck.
i did get OpenGLCD to work but my project is all coded with u8g2 and would love to get this display working.

sonic2wb, 2017-03-25:
Please delete or close I got it to work!! thank u so much for your time and effort!!

olikraus, 2017-03-26:
D/I = dc

closed 2017-03-26 by olikraus