olikraus/u8g2 #227: Issue with DIYMall 1.3" OLED display

smz, 2017-03-26:
Hello and thank you for your library!

I have just started tinkering with a new display I bought on Amazon Italy (https://www.amazon.it/dp/B016HVG0MM/), but it seems there is something wrong and TBH I'm unsure if it's a problem with your library or the display itself: using your u8x8\HelloWorld.ino example (_slightly modified to "fill" the display_) I'm seeing the first character in every line truncated to the left and two columns of apparently "random" pixels on the right side (_see the picture here below_)

[image]

For constructor I used `U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(SCL, SDA, U8X8_PIN_NONE);` as it seemed to me to be the _nearest match_ to my hardware.

Am I doing something wrong or do we have a real issue with that particular display?

I'm at your disposal for whatever test you would like me to do.

Thanks,
Sergio

olikraus, 2017-03-26:
This is a SH 1106 OLED.

smz, 2017-03-26:
Hello, and thanks for answering!

In Amazon description it is declared: `Driver IC: SSD1306`, and they also point to using Adafruit's SSD1306 library...

What constructor do you advice me to try?

smz, 2017-03-26:
... and  (_of course_) you were right:
`U8X8_SH1106_128X64_NONAME_HW_I2C u8x8(U8X8_PIN_NONE);` did the trick!!
Thank you!! 👍

smz, 2017-03-26:
Sorry, one more thing: while I'm set with u8x8, I'm still unable to use my display with u8g2: I can't find what constructor I should use with SH1106 and I2C...

olikraus, 2017-03-27:
Constructor should be:
```
U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
```

Not all constructors are mentioned in the example files...
Full list is here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

olikraus, 2017-03-27:
> Un po' meno luminoso di come me lo aspettavo, ma va benone comunque.

You can try some higher values with the setContast function.

smz, 2017-03-27:
@olikraus Thank-you, your suggested constructor worked perfectly. I'm now wondering what the difference is with the `_VCOMH0_` constructors, as they too seems to work...

**At first sight** tinkering with the contrast doesn't seems to bring benefits: my feeling is that I can reduce the brightness, but not increase it. I'm not yet totally sure about that, thus: I need to make more thorough tests.

If you are interested, I've benchmarked the display with FPS.ino (using my Uno) and the results are:
~~~
Clip=14.5, Clear screen=19.8, @=4.0, Pix=6.7
~~~

As far as regards latin, and having me too a classical studies background (_both Latin and ancient Greek_), I've always maintained it is a really important "bonus" in one's life 👍 

Cura, ut valeas!

Sergio

olikraus, 2017-03-27:
Thanks, for the feedback on the performance.
Yes, the brightness is already very hight by default. VCOMH0 is even lesser bright, but allows a larger brightness range.

smz, 2017-03-27:
Got it!

Thanks again for your beautiful job with this really useful and professional lib. Starred!

You can close this (non-)issue when you wish: everything is OK here now.

closed 2017-04-09 by olikraus