olikraus/u8g2 #228: Runnning Multiple Displays on a Single Processor

Tulsa2008, 2017-03-28:
03/27/17

I am not sure if this is the place to ask a question and I'm sorry if it is not.

Can this library be made to run two displays on a single Processor??

I am running a Mega2560 and need to run two at a time.

Ken

olikraus, 2017-03-28:
Yes, should be no problem.

Tulsa2008, 2017-03-28:
03/28/17
Would you do it Simular to the ADA TFT screens?? 

Like this??
U8G2_ST7565_EA_DOGM128_F_4W_HW_SPI u8g2 (U8G2_R0,  4, 3, 11);
U8G2_ST7565_EA_DOGM128_F_4W_HW_SPI u8g2A (U8G2_R0, 6,  5, 11);

Thanks in advance for your help
Ken

olikraus, 2017-03-28:
Yes exactly. DC line could be shared also.

Tulsa2008, 2017-03-28:
Thank You, Thank You, Thank You,  This will solve a lot of my problems..  
I've been playing with fonts.  I found what I want.

Need to adjust the contrast though,  Haven't figured out the call there yet

Thanks for your help

Ken

Tulsa2008, 2017-03-28:
Found the constant for the contrast setting in u8x8_d_st7565.c.  Reset from 0x016 to 0x01d.

It worked.

As far as I am concerned, you can close this thread now.

Now I'm going to get real serious with exploring this library

Thanks again
Ken

olikraus, 2017-03-28:
Ok, however note, that there is a set contrast function.

Tulsa2008, 2017-03-28:
What is the syntex for that, I have not been able to figure that one out

olikraus, 2017-03-28:
.setContrast( value );

Tulsa2008, 2017-03-28:
I ran into trouble trying to run two displays at the same time.  

The program I'm using is listed below.  Along with three pictures of the results. 

If I run Screen1 by its self, it shows up where it should.  
If I run Screen2 by its self, it shows up where it should.  
If I run both Screens at the same time, nothing shows up on Screen1 and everything shows up on Screen2.

Any ideas??

Any assistance you can provide will be most greatly appreciated.

Ken

[image]

[image]

[image]

` 
#include <U8x8lib.h>
#include <Arduino.h>
#include <U8g2lib.h>
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

//U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2(Rotation  (U8G2_Rx,cs,dc3,reset);   
  U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2(U8G2_R0, 4, 3, 11);	
  U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2A (U8G2_R0, 6, 5, 11);

void setup(void) {
 Serial.begin(57600);
  u8g2.begin();
  u8g2A.begin();
}
 `

Tulsa2008, 2017-03-28:
03/28/17

It looks like the code got cut off.  Here it is again.

Ken

`
#include <U8x8lib.h>
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

//Use U8X8_PIN_NONE if the reset pin is not connected

//SCLK  =  52,  MOSI  =  51 
//U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2(Rotation  (U8G2_Rx,cs,dc3,reset);   
  U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2(U8G2_R0, 4, 3, 11);	
  U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2A (U8G2_R0, 6, 5, 11);

void setup(void) {
 Serial.begin(57600);
  u8g2.begin();
  u8g2A.begin();
 
  pinMode(23, OUTPUT);
  digitalWrite(23, LOW);
}

void loop(void) {
  u8g2.clearBuffer();					
  u8g2A.clearBuffer();          

  u8g2.setFont(u8g2_font_mozart_nbp_tr);  // Medium Font
  u8g2.drawStr(0,10,"Screen 1");  

  u8g2.setFont(u8g2_font_5x7_tr);  // Small Font  Most Used
  u8g2.drawStr(0,30,"Screen 1");  

   u8g2A.setFont(u8g2_font_mozart_nbp_tr);  // Medium Font
   u8g2A.drawStr(0,40,"Screen2 ");  

   u8g2A.setFont(u8g2_font_5x7_tr);  // Small Font  Most Used
   u8g2A.drawStr(0,60,"Screen2 ");  

   u8g2.sendBuffer();
   u8g2A.sendBuffer();

  delay(2000);  
}

`

`

olikraus, 2017-03-29:
There is only one buffer for both display!

Do this:
Clear
Draw 
Send
Clear A
Draw A
Send A

Instead of interleaving the commands like in your code.

Tulsa2008, 2017-03-29:
03/29/17

Thanks for the clarification.  Makes sense.

I was wondering about the buffer issue.  I noticed that the program did not consume as much RAM as I had expected.  

Thanks again for the info.  

I really do like this library.  Seems work very smoothly.  

Haven't tried the Bitmap functions yet but I will be doing that soon.

I am changing an existing program that uses two TFT displays with ten screens each.  I am having to go to GLCDs because you can not see the TFTs in the bright sun. The ADA ST7565 displays can be read very easily in the sun.

Thanks again for all of your help and the library

Ken

----- Original Message ----- 
  From: olikraus 
  To: olikraus/u8g2 
  Cc: Tulsa2008 ; Author 
  Sent: Tuesday, March 28, 2017 9:35 PM
  Subject: Re: [olikraus/u8g2] Runnning Multiple Displays on a Single Processor (#228)

  There is only one buffer for both display!

  Do this:
  Clear
  Draw
  Send
  Clear A
  Draw A
  Send A

  Instead of interleaving the commands like in your code.

  —

__________ Information from ESET Smart Security, version of malware database 15165 (20170328) __________

The message was checked by ESET Smart Security.

http://www.eset.com

Tulsa2008, 2017-03-29:
It worked

Thanks
Ken
  ----- Original Message ----- 
  From: olikraus 
  To: olikraus/u8g2 
  Cc: Tulsa2008 ; Author 
  Sent: Tuesday, March 28, 2017 9:35 PM
  Subject: Re: [olikraus/u8g2] Runnning Multiple Displays on a Single Processor (#228)

  There is only one buffer for both display!

  Do this:
  Clear
  Draw
  Send
  Clear A
  Draw A
  Send A

  Instead of interleaving the commands like in your code.

  —

__________ Information from ESET Smart Security, version of malware database 15165 (20170328) __________

The message was checked by ESET Smart Security.

http://www.eset.com

olikraus, 2017-03-29:
Monochrome displays have indeed advances over TFT in bright sunlight. Look out for transmissive or even better so called reflective glcd s. My favorite is the ea dogs102 glcd as reflective version.

Tulsa2008, 2017-03-29:
3/27/17

The one I am using ADA's Product ID 250 works well outside.  It would be nice if it had a lighter colored background.  The dark green background makes it necessary to run a back light in the house.  

It is very hard to beat the resolution of the TFTs though.  I'll look up the Dogs 102.  

I will be doing another project very similar to this one but it will be in an enclosed car.  I am hoping to use a TFT for that project.  hopefully a bit larger as well.

I tried the procedure you suggested this morning and it worked.  

In this project each of the screens are written as a function.  There are a total of 18 of them.  Some of them are more complicated than others but they are called and run when needed  Only two of them can run at any given time.  Generally, I write the headers for all of the data points only once and  write to the data points that change about once a second.  Using this technique, I will have to write the headers every time I write the data points.  Not to bad as you are using the HW SPI and it is noticeably faster.  Using the software SPI, when you write the entire screen you get a noticeable blink when the screen clears and re-writes.  The TFTs were real bad at this.  Parallel was better than SPI  but neither was good.

I did have a question though, on one of my screens, I have alarms that are printed in rows (lines of text).  Using ADA's library,  I cleared the line by filling it with " " (Blank spaces).  How do you do that with your library??  I tried using the spaces and it did not work.  I guess it is a moot point though since every time I have to write to the screen, I have to write the entire screen.  (Trade-offs).  One buffer = less RAM and that is very important to me in this project.

Thanks for your help.  

Ken 
  ----- Original Message ----- 
  From: olikraus 
  To: olikraus/u8g2 
  Cc: Tulsa2008 ; Author 
  Sent: Wednesday, March 29, 2017 10:12 AM
  Subject: Re: [olikraus/u8g2] Runnning Multiple Displays on a Single Processor (#228)

  Monochrome displays have indeed advances over TFT in bright sunlight. Look out for transmissive or even better so called reflective glcd s. My favorite is the ea dogs102 glcd as reflective version.

  —

__________ Information from ESET Smart Security, version of malware database 15165 (20170328) __________

The message was checked by ESET Smart Security.

http://www.eset.com

olikraus, 2017-03-30:
With u8g2 you have to rewrite everything. However you could use the u8x8 API for direct write to the glcd.

Tulsa2008, 2017-03-30:
That is OK, I like the added feature you have added with would not be available with u8x8.  I will need bitmap.  

Thanks again for the library and you help

Ken

closed 2017-04-09 by olikraus