olikraus/u8g2 #229: Problems Getting Hello World to compile

Tulsa2008, 2017-03-28:
03/27/17

Sorry to bug you but I am having a bit of trouble geeting your library to complie.  
The IDE compiler (linker) seems to be having problems finding things.  I'm sure it just hasn't hound a file where it thinks it should be.  
I have included the ERROR rreport from the compiler.  Also the INO file so you can see what it is looking for.  

I am looking forward to trying this library out.  

Can you provide me some assistance on this??  Any assistance will be greatly appreciated.
Ken Kloster

Compiler Error Report:
D:\1Kloster\1PROJE~1\ARDUIN~1.13\arduino-builder -dump-prefs -logger=machine -hardware D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware -tools D:\1Kloster\1PROJE~1\ARDUIN~1.13\tools-builder -tools D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr -built-in-libraries D:\1Kloster\1PROJE~1\ARDUIN~1.13\libraries -libraries D:\1Kloster\1 Projects\Dashboard\Sketches\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10613 -build-path C:\DOCUME~1\KLOSTE~1\LOCALS~1\Temp\arduino_build_813381 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr -prefs=runtime.tools.avrdude.path=D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr -verbose D:\1Kloster\1 Projects\Dashboard\Sketches\Sketch_3_27_17_U8g2HelloWorld\Sketch_3_27_17_U8g2HelloWorld.ino
D:\1Kloster\1PROJE~1\ARDUIN~1.13\arduino-builder -compile -logger=machine -hardware D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware -tools D:\1Kloster\1PROJE~1\ARDUIN~1.13\tools-builder -tools D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr -built-in-libraries D:\1Kloster\1PROJE~1\ARDUIN~1.13\libraries -libraries D:\1Kloster\1 Projects\Dashboard\Sketches\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10613 -build-path C:\DOCUME~1\KLOSTE~1\LOCALS~1\Temp\arduino_build_813381 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr -prefs=runtime.tools.avrdude.path=D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr -verbose D:\1Kloster\1 Projects\Dashboard\Sketches\Sketch_3_27_17_U8g2HelloWorld\Sketch_3_27_17_U8g2HelloWorld.ino
Using board 'mega' from platform in folder: D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr
Using core 'arduino' from platform in folder: D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr
Detecting libraries used...
"D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\cores\arduino" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\variants\mega" "C:\DOCUME~1\KLOSTE~1\LOCALS~1\Temp\arduino_build_813381\sketch\Sketch_3_27_17_U8g2HelloWorld.ino.cpp" -o "nul"
"D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\cores\arduino" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\variants\mega" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\libraries\u8g2-master" "C:\DOCUME~1\KLOSTE~1\LOCALS~1\Temp\arduino_build_813381\sketch\Sketch_3_27_17_U8g2HelloWorld.ino.cpp" -o "nul"
"D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\cores\arduino" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\variants\mega" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\libraries\u8g2-master" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\SPI\src" "C:\DOCUME~1\KLOSTE~1\LOCALS~1\Temp\arduino_build_813381\sketch\Sketch_3_27_17_U8g2HelloWorld.ino.cpp" -o "nul"
"D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\cores\arduino" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\variants\mega" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\libraries\u8g2-master" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\SPI\src" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\Wire\src" "C:\DOCUME~1\KLOSTE~1\LOCALS~1\Temp\arduino_build_813381\sketch\Sketch_3_27_17_U8g2HelloWorld.ino.cpp" -o "nul"
"D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\cores\arduino" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\variants\mega" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\libraries\u8g2-master" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\SPI\src" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\Wire\src" "D:\1Kloster\1PROJE~1\ARDUIN~1.13\libraries\u8g2-master\U8x8lib.cpp" -o "nul"
"D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\cores\arduino" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\variants\mega" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\libraries\u8g2-master" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\SPI\src" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\Wire\src" "D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\SPI\src\SPI.cpp" -o "nul"
"D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\cores\arduino" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\variants\mega" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\libraries\u8g2-master" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\SPI\src" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\Wire\src" "D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\Wire\src\Wire.cpp" -o "nul"
"D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\cores\arduino" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\variants\mega" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\libraries\u8g2-master" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\SPI\src" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\Wire\src" "D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\libraries\Wire\src\utility\twi.c" -o "nul"
Generating function prototypes...
"D:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\cores\arduino" "-ID:\1Kloster\1PROJE~1\ARDUIN~1.13\hardware\arduino\avr\variants\ [... cut ...]
    U8x8 Text Only Example: No RAM usage, direct communication with display controller. No graphics, 8x8 Text only.
*/

//SCLK  =  52,  MOSI  =  51 
U8G2_ST7565_EA_DOGM128_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 4, /* dc=*/ 3, /* reset=*/ 11);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB14_tr);	// choose a suitable font
  u8g2.drawStr(0,20,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}

olikraus, 2017-03-28:
Is everything working in the Arduino IDE?

olikraus, 2017-03-28:
Did you install U8g2 via normal library​ install menu?

Tulsa2008, 2017-03-28:
03/28/17
Not really,  I tried to use the "install Zip" file method and that didn't work.  IDE came back and said" no Library found" ( I have seen that before).  So I extracted the library from the "ZIP" file to a location where IDE would not see it (for backup purposes) and copied the entire directory " u8g2-master" into the "arduino/libraries" directory.  Restarted IDE and when I went to "Sketch/Include libary", the library is there.  When I go to "files/Example" menu, the examples are there.  It looks like normal.  

When I tried to compile, IDE came back and said it could not find the u8g2.h file.  At that time, I went to the "cppsrc" sub-directory and copied all three of the files to the "u8g2-master" sub-directory.  Its next complaint was that it could not find the "u8g2.h" file, so I copied it from the "Csrc sub-directory to the u8g2-masterfile.

As for the IDE, yes it works, It doesn't like XP very well but most of the time I can get it to work as it should.  

I know the problem is the installation, I'm just not sure where.  I am running IDE version 1.6.13

One thing that might be an issue is that almost everything in my computer runs on "D" drive.  "C" drive is where the operating system is but for the most part, all programs including IDE are installed on "D" drive.  I did notice that IDE uses "C" drive for temporary files.  This machine was setup by Dell and "C" drive is very small (80 gig).  "D" drive is a Terabyte drive.  

If there is anything else I can do to help or info you need, please let me know.

Thanks in advance for your help

Ken

Tulsa2008, 2017-03-28:
3/28/17

Got it running!!

I copied all of the contents of the "csrc" sub-directory into the"u8g2-master" directory and at least  the "HELLO WORLD" sketch works.  

Now I' trying to set the contrast a little darker for the display.  I see there is a "setContrast" function and it needs a uint8 input but at this point, the compiler cant find the function.

closed 2017-03-29 by Tulsa2008