olikraus/u8g2 #231: jlx19264 nothing on display

eddiegeerts, 2017-03-29:
hello, 
I am trying to get a JLX19264 running but the display stays empty.

using U8G2_UC1604_JLX19264_1_4W_SW_SPI on Teensy 3.1 and also tried ESP8266 (also tried HW)
perhaps you can help me with this?

[image]
 
thanxs E. Geerts

olikraus, 2017-03-30:
What about reset signal? It should be high.

eddiegeerts, 2017-03-30:
Hi,yes it is high, I see a little dip when you start the Teensy. Do you have some documentation about this display? I couldn't find anything on www

olikraus, 2017-03-30:
There are several versions, which are called JLX19264

These are the datasheets of the two glcds, which i have in my lab:

http://www.jlxlcd.cn/UpFile/2013122811212.pdf

http://www.jlxlcd.cn/UpFile/2016317111232.pdf

eddiegeerts, 2017-03-30:
thanks for the documents I will have look but I guess is not going to be easy (it will take a while)
just to let you know I have the JLX19264G-333 v1.2  

thanks for the help

closed 2017-03-30 by eddiegeerts

olikraus, 2017-03-30:
I also found something on the the JLX19264-33
[image]

... maybe this helps...

eddiegeerts, 2017-03-31:
no it does not ;) the pinout is not the problem they are described on the pcb so I do not think that is the problem there is something else ...  but what?
Anyway I appreciate the effort if you come up with any suggestions that would be cool
just one thing the doc's talks about initial setup ... a bunch of hex numbers to setup the display... I do not see any of those number on my logic annalyzer also I do not know what they should do because it is in weird language :) I will have a look with google translate perhaps I will find the issue...
I will let know... thanxs

olikraus, 2017-03-31:
The setup code is transfered by u8g2 for this type of display. Indeed you should see that code in your analyzer. However, U8g2 my deviate in the sequence a little bit. I usually do some optimization on the init code itself.

You could
 * send your test code so that i can check the software
 * send a picture of your wiring. I could have a look at this too.

eddiegeerts, 2017-04-01:
great that you want to help thanks a lot! I have attached a zip file with images + code, I hope they help you to find the problem..
[jlx19264.zip](https://github.com/olikraus/u8g2/files/887458/jlx19264.zip)

olikraus, 2017-04-01:
looks good from my perspective, however i was not able to see your board that clearly, so i was not able to check whether for example the yellow wire (clock) is connected to pin 13 (according to your software setup.)
My suggestion is to write down a table with: 
 * Description (clock, data, etc)
 * Board pin number
 * wire color

Then crosscheck whether this is consistent

Oliver

eddiegeerts, 2017-04-01:
U8G2_UC1604_JLX19264_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 

jlx19264   wire    Teensy 3.1
 leda     brown      3.3v
 vdd      red        3.3v
 vss      black      gnd
 sda      orange      11
 sck      yellow      13
 rs       green        9
 rst      blue         8
 cs       purple      10

olikraus, 2017-04-01:
looks perfect... 
slowly i am running out of ideas

eddiegeerts, 2017-04-01:
me too :( anyway if you come up with something that would be great meanwhile I will investigate some more.. thanks a lot !

olikraus, 2017-04-01:
give me a second... there is a difference in the display

olikraus, 2017-04-01:
[image]

is the linear regulator missing on your display?

eddiegeerts, 2017-04-01:
yep  it is not there

olikraus, 2017-04-01:
On my display there is a classic LM1117 (well, probably a low cost clone):

[!http://linhkienvn.com/Uploaded/Members/6363/images/hand-crank-3_3V-generator.png](http://linhkienvn.com/Uploaded/Members/6363/images/hand-crank-3_3V-generator.png)

eddiegeerts, 2017-04-01:
my display has a 0 ohm resistor on R2 so output and input of the LMxxx are connected, anyway I will measure the voltages on the display flatcable just to be sure..

olikraus, 2017-04-01:
ok, makes sense...

eddiegeerts, 2017-04-01:
yes the voltage on the display connector is OK and also the other wires from the board to display seems to be ok..

eddiegeerts, 2017-04-01:
you are not going to believe this.... I bought 2 of these displays and gues what? the second one is working GRRRR
sorry for all of this! anyway thanks for the help and the nice library !!

olikraus, 2017-04-01:
:)

wchpikus, 2018-07-11:
on the datasheet i see 👍 void display_graphic_192x64(uchar *dp)
{
uchar i,j;
for(i=0;i<9;i++)
{
lcd_address(1+i,1);
for(j=0;j<192;j++)
{
transfer_data(*dp);
dp++;
}
}

so, it can..
how to send bmp?
u8g8.drawXBMP( 0, 0, welcome_screen_width, welcome_screen_height, welcome_screen); = 'u8g8' was not declared in this scope

olikraus, 2018-07-11:
You use the Arduino version of u8g2? If so, there are many examples for U8g2 available in the examples menu of the Arduino IDE.

wchpikus, 2018-07-11:
2.22.18 arduino 1.8.5
my lcd works, i can write some text,
e.g 
void display_status(){
        local_ip = WiFi.localIP();
        u8x8.setFont(u8x8_font_chroma48medium8_r);
        u8x8.draw1x2String(0,0,"LCD ip:");
        u8x8.setCursor(0, 3);
        u8x8.print(local_ip);
              
}

olikraus, 2018-07-11:
U8x8 is the low level text only interface.

wchpikus, 2018-07-11:
Ok, so i should use this second?
When i switch to this one, lcd no respond (offcource u8x8 replaced to u8g2 commands).
What is the first parameter of U8G2_UC1604_JLX19264_F_4W_SW_SPI u8g2? 
I will try later, now is to late for test.

wchpikus, 2018-07-13:
ok with u8gt works well:)
tnx for quick response:)