olikraus/u8g2 #232: SSD1306 128x64 display gets clipped when using the 1 or 2 buffer size

JanHolbo, 2017-03-29:
Thank You for a great library!

I am writing a sketch for a atmega328 using a I2C SSD1306 128x64 OLED and it works fine with the Full Frame Buffer constructor (U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);)

I would like to preserve RAM and use preferably the small (1) framebuffer or even the (2) framebuffer as the full framebuffer leaves just 318 bytes for local variables according to the compiler. And I am far from having completed this sketch.

But using (U8G2_SSD1306_128X64_NONAME_2_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);) results in just the lower 3rd or 4th of the screen to be displayed (and even half that when using _1_HW).

I have tried using VCOMH0 instead of NONAME - same result.

I am wondering if this is user error or a bug?

Jan

olikraus, 2017-03-30:
For _1_ or _2_ constructor s, you must not use the send command any more. Instead use the first page/next page loop. See the examples for this.

JanHolbo, 2017-03-30:
Thank You so much. User error. Note to self: Read The Flippin' Manual  (RTFM)

closed 2017-03-30 by JanHolbo