olikraus/u8g2 #233: Font memory size request

picclock, 2017-03-31:
Hi Olikraus
Thanks for an amazing library, almost impossible to use displays well without it. However as you are aware memory sizes on these micros are small. Would it be possible for you to put the memory size of the Fonts on your selection list so we can balance our creative artistic display needs whilst leaving enough room for the program to run.

Many Thanks 

Best Regards

picclock

olikraus, 2017-03-31:
Which selection list? The font size is always mentioned in the "font overview image".
Moreover on the wiki pages, only the font name itself is mentioned, like "6x12" font, however this font itself has several "sizes" depending on the number of glyphs included:

https://github.com/olikraus/u8g2/wiki/fntgrpx11#6x12

JamesGKent, 2017-03-31:
i think this question is asking for some way to check how much memory each font takes up on the micro, as opposed to pixel sizes of the glyphs.

in which case you are right, they are listed, just not in the overall section as per your last link. however given the size varies with the number of glyphs, it would not be practical to include it there.

picclock, 2017-03-31:
Hi
Thanks for the prompt response. I was using the list at https://github.com/olikraus/u8g2/wiki/fntlistall, and was unaware of the comprehensive list you have just indicated. FWIW it may be useful to put a further information link to that page on the page I used. 

Thanks for the information, and an excellent library. 

Best Regards

picclock

olikraus, 2017-03-31:
Hmm... i just relized, that one of the tables on https://github.com/olikraus/u8g2/wiki/fntlistall
is incorrect.

picclock, 2017-03-31:
Hi 
I have also just realised that the page you directed me to only lists the first few fonts with details - so perhaps my first request could still stand.

Best Regards

picclock

olikraus, 2017-03-31:
picclock, 
there is a overview picture for EACH font. This picture includes the size of the font in the sense of occupied bytes in the uC. In fact i do not know of any missing picuture, but if you found a font without picture, please let me know.

>  I have also just realised that the page you directed me to only lists the first few fonts with details

Maybe you can give some details here.

picclock, 2017-03-31:
Hi

Yes I see now, you have to select the font name link to reveal the details.

Many thanks for your prompt attention.

Best Regards

picclock

closed 2017-03-31 by picclock

olikraus, 2017-03-31:
yes, exactly.