olikraus/u8g2 #234: [Font Request] Farsi font

information-security, 2017-03-31:
Thanks for the great job done so far. Can you please embed a farsi/persian font as well? or let me know how can I add some more custom fonts?

Thanks

olikraus, 2017-03-31:
What unicodes do you required? I need a list of unicodes for adding this.

You may also look here. Will the arabic font include your glyphs also?

For some arabic fonts, please look here
https://github.com/olikraus/u8g2/wiki/fntgrpx11#10x20
and also here
https://github.com/olikraus/u8g2/wiki/fntgrpunifont

information-security, 2017-04-03:
Well, Arabic language generally doesn't have four characters of Farsi language (پ ژ گ چ Unicode: [067E, 0698, 06AF, 0686]). But the font in your second link seems to have those four character included!

olikraus, 2017-04-09:
ok.. closing this...

closed 2017-04-09 by olikraus

information-security, 2017-04-10:
Sorry for commenting again on closed issue. I couldn't get it working. Have a look at following code, it just doesn't print anything on the 126x64 GLCD.

```
u8g2.clearBuffer();          // clear the internal memory
u8g2.setFont(u8g2_font_unifont_t_arabic);
char farsi[] = "سلام";        // Word hello written in farsi
u8g2.drawStr(5,5, farsi);
u8g2.sendBuffer();           // transfer internal memory to the display
```
Thanks

olikraus, 2017-04-11:
Code looks correct. Can you print Latin glyphs?

information-security, 2017-04-11:
Yes latin texts are printed correctly!

information-security, 2017-05-01:
Hi,
Any update on this?

olikraus, 2017-05-01:
oh, excause me, i just saw the mistake.
You must use 
u8g2.drawUTF8(5,5, farsi);

information-security, 2017-05-01:
Wow, thanks. It's working now but another problem came up...
As you can see in farsi, characters may attach to each other. For example the characters (From right to left) س ل ا م shape the word سلام. All of them are attached except last one. Now when I print سلام it is mistakenly printed as م ا ل س on LCD. Words are not attached and are in reverse order as well.

olikraus, 2017-05-01:
yes, this is the kerning problem. However I decided, that u8g2 does not support kerning. For arabic font the kerning table will be very huge (100KB or so if i remember correctly) which does not make sense for an embedded system. However there is a workaround, which is described here:

https://github.com/olikraus/u8g2/issues/42

information-security, 2017-05-01:
I went through the discussion in #42. With kerning table your are reducing the space between characters, am I right? If so, I think there would be a better solution because kerning isn't attaching chars together it's just making them closer to each other.
Let's make it simple. Just take first character of the word سلام as an example. u8g2_font_unifont_t_arabic has the س char with value of 0x0633 and other required printable shapes of this char are also available. Take these scenarios:
1) If س is occurring at the end of a word then it must be printed same as it does now (such as: ترس).
2) If س is occurring at the beginning of a word then 0xFEB3 must be printed (such as: سلام)
3) If س is occurring at the middle of a word and the char before the س is an attachable char then 
0xFEB4 must be printed (such as: پسته)
4) If س is occurring at the middle of a word and the char before the س is NOT an attachable char then 0xFEB3 must be printed (such as: پرستار)

A few examples of NON-Attachable chars when they occur before another char: (ر ز و د ذ)

I think we can come up with a solution by doing a few lines of code and no kerning table would be required at all. Let me know your idea.

olikraus, 2017-05-01:
Nice explaination. However, i think there is nothing much what i can doo. 0xFEB2 to 0xFEB4 are both part of the arabic font. So you can just print this instead of 0x0633. It is, so to say, your responsibility to print the correct glyph.