olikraus/u8g2 #235: SSD1322 u8g2.getWidth() reports 240 pixels, leaving a black bar at the end.

DanielWittenaar, 2017-04-01:
I upgraded today to the u8G2 lib to have a faster refresh of the screen. 

When using the SSD1322 driver i get a black bar at the end.  Also with the test programs it looks like it screws up some memory mapping.  I see some tailings in the area were the black bar is.

I'm using an Arduino M0 ARM on Full Buffer mode. 
LCD is a SSD1322  from alibaba 
Line enabled = 
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	// Enable U8G2_16BIT in u8g2.h

When i add this line to any example program:
SerialUSB.print("  Width  ");SerialUSB.print(u8g2.getWidth()); SerialUSB.print("  Height  "); SerialUSB.println(u8g2.getHeight());
It reports back:  Width  240  Height  64

So i did check the library to see if i could find the source of this error..  

Added note::
Did a test with a moded Leonardo on 3.3V   With Page buffer and had the same result Width 240 and a black bar..

Tested it with the old library:
U8GLIB_NHD31OLED_GR u8g(10, 9 , 8);  //  ChipSelect ,  D/C , reset

And it report a Width of 248     Which is 8 more then the new lib..  But still 8 are missing.. 

How do i get the library to use the whole 256 pixels?

Greets Daniel

olikraus, 2017-04-01:
Indeed strange. Are you sure, that all sources are recompiled after changing to 16bit mode?

DanielWittenaar, 2017-04-02:
I did not recompile anything.   
Using Arduino 1.8.2  from Arduino.cc  Gone to the "library manager" and added the U8g2 lib V2.14.7

Then opened up the GraphicsTest example..  Notished it had some things going on at the right of the screen.  And notished the black bar at that moment.
So added the line to check for the resolution and came up with 240 Width..   
So then i got back to the old lib on a leonardo and found that it also had a black bar but now on the beginning and 8 pixels smaller.. (Never notished that before as i only had Font Size 30 text on it.. So i thoughed that was the lead in from the font..   )

Thanks in advance, Daniel
[image]
[image]
[image]
[image]
[image]

olikraus, 2017-04-02:
Did you uncomment U8G2_16BIT in u8g2.h?
If not please do so and repeat your tests.

DanielWittenaar, 2017-04-02:
Thanks for your fast response.

Indeed the U8G2_16BIT was uncommented..  #RTFC > Read The F Comments...  
Been playing for an hour or so to get the library to auto detect if resolution is above 240 to go in 16 bit mode..  And tryed to integrate the 16 bit mode into the LCD selection line..   Had no quick succes so i leave it by this and need to remember to put it on or off when compiling code for a different display..

Thanks for the Support and great work on the library!! Happy Coding!

closed 2017-04-02 by DanielWittenaar