olikraus/u8g2 #236: DrawBox with 0 Height shouldn't display anything

vbbartlett, 2017-04-02:
drawBox(x,y,w,0) will draw a box at a boundary line.  I would expect the h = 0 to not draw anything.

    u8g2.setFontMode(1);
    u8g2.setDrawColor(1);
    u8g2.drawBox(40, 18, 13, 0);

This is using U8G2_PCD8544_84X48_1_4W_SW_SPI u8g2(U8G2_R0,...)
On a nokia 5110

I have also noticed that drawFrame doesn't always draw the box correctly.  It will sometimes leave off the bottom if the size of the box crosses a boundary line.  I have also seen it flicker on and off past the bottom of the boundary line.

I am using firstPage with the do while loop

olikraus, 2017-04-02:
You have a complete example for the frame problem?

vbbartlett, 2017-04-02:
Yes, it is a full project, so I will send it all and point you to the lines that seem to be problematic. 
It will be a later today or tomorrow.  Thanks.

olikraus, 2017-04-02:
Maybe you can reduce the problem so that I can see the issues with the frame. The zero height for the box is certainly a problem.

vbbartlett, 2017-04-02:
here is the code
Please note that my wiring is not the same, so I would replace that line.  (I used that because i have other projects that use lcd5110 graph and is wired that way.

```
#include <U8g2lib.h>
U8G2_PCD8544_84X48_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 11, /* dc=*/ 10, /* reset=*/ 12);  // Nokia 5110 Display
char pump[2]  = {0x62,0};
char active[2] = {0x52,0};
char inactive[2]= {0x51,0};

void setup() {
  // put your setup code here, to run once:
 u8g2.begin();
}
int frameCnt = 0;
void DrawWellStatus(bool on, bool act, int sec)
{
  int h = 18;
  int l = 13;
  int x = frameCnt %8;
  u8g2.setFont(u8g2_font_m2icon_7_tf);
  u8g2.drawStr(l+33, h, on?active:inactive);
  if (on && act)
  {
    u8g2.setFontMode(1);
    u8g2.setDrawColor(1);
    u8g2.drawStr(l+40, h, pump);
    u8g2.setDrawColor(2);
    u8g2.drawBox(l+40, h+x+10, 12, 8);
    u8g2.drawFrame(l+20, h+x+10, 12, 8);
    u8g2.drawBox(l+40, h-x, 12, 8);
    u8g2.drawFrame(l+20, h-x, 12, 8);
    u8g2.setDrawColor(1);
    frameCnt++;
  }

  u8g2.setFont(u8g2_font_courB08_tr);
  u8g2.drawStr(l,h,"Well");
}

void loop() {
  // put your main code here, to run repeatedly:
 u8g2.firstPage();
  do {
    u8g2.drawFrame(0,0,84,10);
    u8g2.drawFrame(0,0,84,48);
    DrawWellStatus(true, true, 10);
  } while ( u8g2.nextPage() );
  delay(1000);
}
```

olikraus, 2017-04-07:
ok, thanks, i will have a look at this

olikraus, 2017-05-14:
Hmmm... frameCnt is modified within the firstPage/nextPage loop. This will indeet not work and makes the behavior unpredictable.

olikraus, 2017-05-14:
I made drawBox more robust. For the other problem: As long as frameCnt is updated within the loop, it can not work.
Feel free to reopen this issue...

closed 2017-05-14 by olikraus