olikraus/u8g2 #237: DrawXBM issues

Tulsa2008, 2017-04-02:
04/01/17

For the most part , I have been able to get everything I have tried in your library to work but DrawBMX is giving me a bit of trouble.  I am using an ADA Product 250 ST 7565 display with the full buffer option.  The processor is an Arduino Mega.  The display(s) I have two work fine.

When I tried to use the DrawBMX function, I was able to draw lines by setting up an array with alternating rows of OxFFs and Ox00s. I got what I expected.  Multiple rows of lines across the screen.

When I tried to display my bit maps generated from crude BMP files using the "Image2cpp" conversion program, I get a display that kind of looks like my BMP but not quite.  The image kind of looks like it might be mirrored.  The image by the way is 122X56 on a 128X64 display.  

What are you using (or can suggest) to convert  BMP files to XBM files that will work.  I have to believe that I am missing something basic.   

If it would help, I could send you a picture of the display.  I have used Image2cpp with ADA's library and it works although their "DrawBitMap" function for the ST7565 has significant problems.

Any light you can shed will be greatly appreciated.   
Ken

olikraus, 2017-04-02:
GIMP generates the correct image format.

Tulsa2008, 2017-04-02:
Thanks for your input.  I'll look that one up.

Ken 
  ----- Original Message ----- 
  From: olikraus 
  To: olikraus/u8g2 
  Cc: Tulsa2008 ; Author 
  Sent: Saturday, April 01, 2017 10:06 PM
  Subject: Re: [olikraus/u8g2] DrawXBM issues (#237)

  GIMP generates the correct image format.

  —

__________ Information from ESET Smart Security, version of malware database 15187 (20170401) __________

The message was checked by ESET Smart Security.

http://www.eset.com

tongbajiel, 2017-04-02:
Hello Olikraus ...
Can you build this library for me, to be use with PCF8814 nokia 1100 screen. 
I was confused to modify itself many times.
Sorry, my English is bad. I come from Indonesia
Thank you in advance.

Tulsa2008, 2017-04-04:
Thanks for the hint.  Gimp is a bit more tool than I need but , Hey, it's always nice to have a nice tool to work with.

Thanks again
Ken

olikraus, 2017-04-04:
:)

closed 2017-04-09 by olikraus

fire1, 2017-04-09:
Please add this information into WiKI method's docs (like a note ) Thanks!

olikraus, 2017-04-16:
@fire1 GIMP isnow mentioned.