olikraus/u8g2 #240: Request support for PCF8814 Nokia 1100 display

tongbajiel, 2017-04-04:
Hello Olikraus ...tanks lot for this great library.
My LC7981 LCD from old panasonic PBX telephone now can be used with this lib.

Can you add support for PCF8814 nokia 1100 screen, with this connection (SCLK, SDA, SCE/EN, RESET) in arduino.
I was confused to try modify from PCF8812 to PCF8814 itself many times.
This two option cant be used for me.

//U8X8_PCF8812_96X65_4W_SW_SPI u8x8(/* clock=/ 13, / data=/ 11, / cs=/ 10, / dc=/ 9, / reset=*/ 8);	// Could be also PCF8814

//U8X8_PCF8812_96X65_4W_HW_SPI u8x8(/* cs=/ 10, / dc=/ 9, / reset=*/ 8);						// Could be also PCF8814

Sorry, my English is bad. I come from Indonesia
Thank you in advance.

olikraus, 2017-04-04:
What's the problem with the existing drivers?

tongbajiel, 2017-04-06:
On pcf8814.. two default option that set for pcf8812, cannot be used.
pcf8814 (nokia 1100 lcd) does not use "dc" pin.
Only SCK/clock, SDIN/sda, SCE, and RST that available in lcd pinout.

olikraus, 2017-04-09:
ok, i guess adding the 3-wire interface will solve this problem

tongbajiel, 2017-04-09:
Tanks before... i will try it and report ASP.

olikraus, 2017-05-20:
ok, i think this is done

closed 2017-05-20 by olikraus