olikraus/u8g2 #241: ST7565_NHD_C12864 seem to not work

hieuvt6-fti, 2017-04-05:
I have tried both `U8G2_ST7565_NHD_C12864_1_4W_HW_SPI` and `U8G2_ST7565_NHD_C12864_1_4W_SW_SPI`, neither worked. In particular:

    //U8G2_ST7565_NHD_C12864_1_4W_HW_SPI u8g2(U8G2_R0, 10, 9, 8);
    U8G2_ST7565_NHD_C12864_1_4W_SW_SPI u8g2(U8G2_R0, 13, 11, 10, 9, 8);
    
    void setup()
    {
      u8g2.begin();
    }

    void loop()
    {
      u8g2.firstPage();
      do {
        u8g2.setFont(u8g2_font_ncenB14_tr);
        u8g2.drawStr(0,22,"Hello World!");
      } while ( u8g2.nextPage() );
      delay(1000);
    }

I have also tried the equivalent LCD support in U8glib. 

    U8GLIB_NHD_C12864 u8g(13, 11, 10, 9, 8);  // SPI Com: SCK = 13, MOSI = 11, CS = 10, A0 = 9, RST = 8

U8glib worked.

Of course, I maintain the same hardware for all attempts. CLK at 13, MOSI at 11, CS at 10, A0/DC at 9 and RST at 8.

olikraus, 2017-04-08:
There was an issue with the contrast setting for this device in U8g2. 
Can you apply u8g2.setContrast(180) or even higher values?

olikraus, 2017-05-16:
no further answer... closing this. Please reopen if required.

closed 2017-05-16 by olikraus