olikraus/u8g2 #242: Issue with different "User Pages"

FelixMue, 2017-04-07:
Hi Oli,

i'm not so sure if this is the right way, but i didn't find any other chance to get in touch with you. I'm curious how to get different "User Pages" working but couldn't really find something promising. The thing i try to do is a User Interface which displays various sensor data such as acceleration, speed, etc.. Cause this is a lot of data the text is pretty small and hard to read in detail (if not directly in front of the display). My solution to this would be an "overview" page and 2-3 pages with bigger, detailed information which can be scrolled through via button press. How can i get this with u8g2 and/or M2tklib?

maybe you can drop an email adress here to discuss this via PM?

cheers

Felix

olikraus, 2017-04-07:
Probably this is more a programming​ question, than a U8g2 question.
But first of all you should decide what exactly you want: scrolling or pages.

FelixMue, 2017-04-07:
should be pages.

i found this on yt https://www.youtube.com/watch?v=wtU02eBNJfQ 
i want to do the same he does at about 4:50.

olikraus, 2017-04-08:
Six displays on six different uC with six different outputs? I think that person just wrote six different ino files for that. Maybe you should ask the author of the video.

FelixMue, 2017-04-08:
i want to do this with only one oled. i don't know how to code this "page switch" on button press. more a programming question, i know.

thanks for your help

closed 2017-05-04 by olikraus