olikraus/u8g2 #243: u8x8_d_ssd1309 default_x_offset

imarinl, 2017-04-08:
Default_x_offset in this static const u8x8_display_info_t u8x8_ssd1309_128x64_noname_display_info is setting to 2. So all pixel are shift right 2 positions. It's better setting to 0

olikraus, 2017-04-09:
yes, indeed... but i wonder why i have not seen the pixel offset here:

https://github.com/olikraus/u8g2/wiki/gallery#13-jan-2017-ssd1309-128x64

olikraus, 2017-04-09:
ok, for the display I own (see link above), this offset is required. This probably means, the display i have is not a SSD1309, but something else.

Will your display work with setting the pixel offset to 0?

olikraus, 2017-04-09:
introduced a noname0 device, renamed the noname device to noname2:

//U8G2_SSD1309_128X64_NONAME0_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
//U8G2_SSD1309_128X64_NONAME0_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
//U8G2_SSD1309_128X64_NONAME2_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
//U8G2_SSD1309_128X64_NONAME2_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

imarinl, 2017-04-11:
Mi OLED is UG-2864KLBMG01 with driver SSD1309.
I'm running the library in a linux environment, not Arduino.
0 is ok for me. u8glib was ok, and reviewing hexadecimal packets y found differences in the offset sended.
The solution given i think it's ok.
With offset 0 all perfect for me.
Thanks for the library, it is so useful.

olikraus, 2017-04-12:
great, thanks for the feedback

olikraus, 2017-05-19:
ok, i think this is done, closing...

closed 2017-05-19 by olikraus