olikraus/u8g2 #244: Method to specify specific hardware SPI interface

Industruino, 2017-04-10:
Dear Olikraus,

Thank you for creating this wonderful library! I would like to request support for a method to define which hardware SPI interface is used to drive the LCD. We created a board where the LCD is connected to a secondary hardware SPI interface, keeping the primary SPI interface clear for other user applications. A convenient way to specify which SPI interface is used for the LCD would be very useful.

Thank you.
Loic

olikraus, 2017-04-10:
Yes, will be done. I had a similar request recently.

olikraus, 2017-05-04:
Note for myself: Is this implemented already? Analyse Arduino naming convention for other ports.

olikraus, 2017-05-19:
Wire.h looks like this for SAM
```
#if WIRE_INTERFACES_COUNT > 0
extern TwoWire Wire;
#endif
#if WIRE_INTERFACES_COUNT > 1
extern TwoWire Wire1;
#endif
```

SPI.h looks like this:

```
#if SPI_INTERFACES_COUNT > 0
extern SPIClass SPI;
#endif
```

Implementation for Wire1 looks like this
```
/* define U8X8_HAVE_2ND_HW_I2C if the board has a second wire interface*/
#ifdef WIRE_INTERFACES_COUNT
#if WIRE_INTERFACES_COUNT > 1
#define U8X8_HAVE_2ND_HW_I2C
#endif
#endif
```

olikraus, 2017-05-20:
Loic, do you follow the Arduino spec? I mean, in your case, SPI_INTERFACES_COUNT would be true, and the secondary interface is SPI1, right?

olikraus, 2017-05-20:
implemented...
The string "2ND_" will prefix 4W_HW_SPI:

U8X8_ST7565_EA_DOGM128_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8X8_ST7565_EA_DOGM128_2ND_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

closed 2017-05-20 by olikraus

elmarfaber, 2018-12-20:
Dear Olikraus,

is it possible to use the second SPI interface for the ST7920? Combined with the mySensors library I can't use the first SPI. The u8g2lib with the STM32F103C8T6 and first hardware SPI without mySensor library works very good, the software spi doesn't work for me I don't know why.

Best regards

Elmar Faber

olikraus, 2018-12-20:
created a new issue #767 instead...