olikraus/u8g2 #245: Sleep - ST7920

davixete, 2017-04-12:
Hello oliver

I'm trying to turn off a GLCD with ST7920. Connected as SPI. I have tested 'setPowerSave (1)'. Does nothing. How to turn off the GLCD (SLEEP).

a greeting.

PS: I feel googles translate

olikraus, 2017-04-15:
ST7920 does not support power down mode

davixete, 2017-04-15:
thanks oliver!

davixete, 2017-04-18:
Hello oliver again!

I have found that I have to send the following code. To turn off the display.

0b00001000, or in hexadecimal 0x08.

It would be possible to order through: u8g2.print ();

Thanks for the help.

link: http://forum.arduino.cc/index.php?action=dlattach;topic=470771.0;attach=208840

olikraus, 2017-04-18:
This is already done:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7920.c
line 73
The 0x08 command is just a display off.
To use this feature, use the setPowerSave member function.

davixete, 2017-04-18:
Which builder should I use? For the ST7920 there are none.

I have tried several without success.

`U8G2_ST7920_128X64_1_SW_SPI u8g2 (U8G2_R0, 52, 51, 53 / *, reset = 8 * /);` // It does not work

Thanks oliver

davixete, 2017-04-18:
Can it be by the SPI connection?

olikraus, 2017-04-18:
no, it should be independent from the interface.

davixete, 2017-04-18:
thanks oliver!

SAM33123, 2018-09-16:
In the extended instruction set you can turn on stand by mode.

> Normal mode (450uA Typ VDD=5V)
Standby mode (30uA Max VDD=5V) 

>U8X8_C(0x034),            			/* 8 Bit interface (DL=1), extended instruction set (RE=1) */
  U8X8_C(0x001),		                /* Standby mode */

> Enter stand by mode, any other instruction can terminate 

on page 17: [https://www.lcd-module.de/eng/pdf/zubehoer/st7920_chinese.pdf](url)

olikraus, 2018-09-16:
hmm... 

Add
```
U8X8_C(0x034), /* 8 Bit interface (DL=1), extended instruction set (RE=1) /
U8X8_C(0x001), / Standby mode */
```
into u8x8_d_st7920_powersave1_seq()

Wake up does not require any extra command.

... should be tested.

olikraus, 2018-09-16:
thanks, nice finding...

AnHardt, 2018-09-16:
Not related to this problem.
The [ST7920 datasheet V4.0](http://www.hpinfotech.ro/ST7920.pdf) is the latest i know about. Many things are updatet compared to the [V3.0](https://www.lcd-module.de/eng/pdf/zubehoer/st7920_chinese.pdf) liked above. For example they admit, SPI CS is not working - don't use together with other devices on the same SPI buss.

olikraus, 2018-10-28:
it works, but the display will be activated as soon as anything is written to the display.
This means, that actually `u8g2.setPowerSave(0);` is useless.

Anyhow, `u8g2.setPowerSave(1)` will put the device into sleep mode.

closed 2018-10-28 by olikraus