olikraus/u8g2 #247: deatialed C initialization or hello world

khanrezwan, 2017-04-15:
I was trying to use this lib with AVR Atmega328p. But when I was trying call setup, I was lost in the documentation how to start. for example:
`void u8g2_Setup_ssd1306_128x64_noname_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);`

I can find rotation but what is this `u8x8_msg_cb byte_cb` and `u8x8_msg_cb gpio_and_delay_cb`. I can understand the last param has something to do with gpio setup. also lost in `u8x8_byte_4wire_sw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);`
The documentation is cryptic. Please provide few hello world for C

olikraus, 2017-04-16:
There is a wiki page on this:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

There are also some examples for other uC which can be useful.

Also have a look at the video here: https://www.youtube.com/watch?v=MipOGBStBbI

olikraus, 2017-05-04:
closing this...

closed 2017-05-04 by olikraus