olikraus/u8g2 #249: SH1106_128X64_NONAME horizontal and vertical offset with rotation 2

D3add3d, 2017-04-21:
Using U8G2_SH1106_128X64_NONAME_1_4W_SW_SPI u8g2(U8G2_R0, 13, 11, 10, 8, 7); for initialization works fine when rendering, [0,0] is at the top left of the display, however when using R2 the [0,0] is offset.

**R0**
[image]

**R2**
[image]

olikraus, 2017-04-21:
Maybe this is a ssd1306 display

D3add3d, 2017-04-21:
Haven't tried treating it as SSD1306 because I've used some other libraries for SSD1306 and they did not work correctly, libraries for SH1106 did however... I will give it a try, one moment.

D3add3d, 2017-04-21:
Works like a charm, thank you! My apologies, I should have realized my mistake earlier.

closed 2017-04-21 by D3add3d

olikraus, 2017-04-21:
No problem