olikraus/u8g2 #250: support SSD1331 (96x64, 16-bit color)

Jorgen-VikingGod, 2017-04-23:
Hi @olikraus do you think it is possible to support also the OLED display driver SSD1331?
There is currently only one library out there:
https://github.com/adafruit/Adafruit-SSD1331-OLED-Driver-Library-for-Arduino

I ordered my display on [aliexpress](https://www.aliexpress.com/item/0-95-Inch-SPI-Full-Color-OLED-Display-DIY-Module-96x64-LCD-For-Arduino-SSD1306-Driver/32790785282.html?spm=2114.13010208.99999999.262.c5lCiv)

It is a 0.96" 16-bit Color 96x64 pixel OLED display with SSD1331 driver chip over SPI.

olikraus, 2017-04-24:
I have redesigned u8g2 as a pure monochrome library. SSD1331 is supported by ucglib.

Jorgen-VikingGod, 2017-04-24:
Thx for fast reply and sorry for this issue - it was my fault; I did not see your other lib.
Thought it was the same - but older version.

I will close this issue again and will try to use the ucglib.

Thank you very much for your great work and support

closed 2017-04-24 by Jorgen-VikingGod