olikraus/u8g2 #251: Examples: Scrolling Text

scrapes, 2017-04-23:
The Example Scrolling Text isn´t working.
I made my own Version, with Serial input at the beginning. I only could test it on an ESP8266-01 but i think it is working with Arduino´s too.

`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SSD1306_128X32_UNIVISION_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 2, /* data=*/ 0, /* reset=*/ U8X8_PIN_NONE);

u8g2_uint_t width;			// pixel width of the scrolling text (must be lesser than 128 unless U8G2_16BIT is defined
const char *text = "I think of you! Call me ";	// scroll this text from right to left
uint8_t StartOffset = 5;

const uint8_t* font = u8g2_font_helvB24_tr;

const byte numChars = 128;
char receivedChars[numChars]; // an array to store the received data

bool newData = false;

void recvWithEndMarker() {
 static byte ndx = 0;
 char endMarker = '\n';
 char rc;
 
 // if (Serial.available() > 0) {
 while (Serial.available() > 0 && newData == false) {
 rc = Serial.read();

 if (rc != endMarker) {
 receivedChars[ndx] = rc;
 ndx++;
 if (ndx >= numChars) {
 ndx = numChars - 1;
 }
 }
 else {
 receivedChars[ndx] = '\0'; // terminate the string
 ndx = 0;
 newData = true;
 }
 }
}

bool showNewData() {
 if (newData == true) {
 Serial.println("NewData");
 Serial.println(receivedChars);
 newData = false;
 return true;
 }
 return false;
}

void InitText()
{
  while(!showNewData()){
  Serial.println("Please Enter Things to say:");
  recvWithEndMarker(); 
  delay(1000);
  }
  text = receivedChars;
  Serial.println("Text is:");
  Serial.println(receivedChars);
  u8g2.setFont(font);
  width = u8g2.getStrWidth(receivedChars);  // calculate the pixel width of the text
  Serial.println(width);
}

void setup(void) {
	
  Serial.begin(115200);
  Serial.println("START");
  
  u8g2.begin();  

  InitText();
  u8g2.setFontMode(0);		// enable transparent mode, which is faster
}

int loops = 0;
void loop(void) {
  int offset = 0;
  while((offset - StartOffset) < width)
  {
    u8g2.clearBuffer();          // clear the internal memory
    u8g2.setFontMode(0);
    u8g2.setFont(font);
    u8g2.drawStr((StartOffset-offset),26,receivedChars);  // write something to the internal memory
    u8g2.sendBuffer();          // transfer internal memory to the display 
    offset+=2;
    delay(10);
  }
  
  loops++;
  
  if(loops >2)
  {
      InitText();
      loops = 0;
  }
}
`
Only downside: it is Full Buffer Rendering.
Also it only worked for me in the 16-bit mode. 
If you leave out the delay, the microprocessor crashes. 

Side question, why exactly is the 16-bit mode not enabled by default?

JamesGKent, 2017-04-24:
the 8 bit mode is the default due to avr cores being able to do 8 bit maths a lot quicker than 16 bit, by only enabling 16 bit maths when the displays actually require it the result is a library that is quicker for the majority of people, but still fully capable of driving larger displays.

olikraus, 2017-04-26:
@JamesGKent thanks for the precise answer.

@scrapes I will have a look at the example.

olikraus, 2017-05-14:
I had a look at this example.
I think offset and StartOffset should be of type u8g2_uint_t (16 bit unsigned).
Actually the example is too complicated: I can not see whether there is something wrong in u8g2 or whether there is a bug in your code. At least my own scrolling example is working nicely.

olikraus, 2017-05-20:
no further reply...closing...

closed 2017-05-20 by olikraus