olikraus/u8g2 #252: Custom Fonts - Dealing with characters spacing

ReVl27, 2017-04-28:
Dear Olikraus,
I'm studying your u8g2 library, so that I can do some modifications for my project. Because I only need to display text and I'm running out of Flash memory. So I need to reduce/rewrite some function.

My project need a custom font (Agency FB), so I use LCD Image Converter to convert that font into a source file. All glyphs in this font have the same height, however some character, for example "W" is wider "i". I wonder how can you dealing with these different width when you print a string on the screen, so that all character spacing looks equal?

ReV.

olikraus, 2017-04-28:
The font *must* be available as BDF file. The BDF file will include the width information for each gylph. The BDF file is then converted into the internal font format. See also the u8g2 faq:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt

ReVl27, 2017-04-28:
Dear Olikraus,
I downloaded otf2bdf and tried installing FreeType2 Library (GnuWin32) but still I got the error "freetype6.dll is missing".
Is there any simpler method to convert an windows font (.ttf) to .c source code with glyphs width?
ReV.

ReVl27, 2017-04-28:
P/S: I found a utility name GLCD Font Creator.
It has a more user friendly interface and it convert most of the fonts directly to .c code, glyph's width is also included.
Thanks anyway!

olikraus, 2017-04-28:
> directly to .c code

which can not be used with u8g2. I just can only repeat myself: The font must be available as BDF file. Only the u8g2 converter generates the *correct* c code.

ReVl27, 2017-04-29:
OK, I got it.
Otf2bdf can work now and after convert BDF to internal .c using "bdfconv", everything is displaying just fine.
Thank you Olikraus.

olikraus, 2017-04-29:
:)

closed 2017-05-04 by olikraus