olikraus/u8g2 #253: Add 7 Segment Font

koenjer, 2017-04-28:
Hello,
Is it possible to add a 7-segment libary around 30pixels high?
Font: http://www.dafont.com/lcd-lcd-mono.font
thanks.

olikraus, 2017-04-28:
Could be done, but I do not think it will look good

koenjer, 2017-04-29:
[digitalnumbers-68.zip](https://github.com/olikraus/u8g2/files/966015/digitalnumbers-68.zip)

Hi Oli,

It is used for numbers only. I adjusted the bitmap in fontforge so it look ok.
Could really appreciate if you could add this. Height around 68.

Grtz.

olikraus, 2017-04-29:
I have used the following command:
./bdfconv -v -f 1 -m  '32,48-57' digitalnumbers-68.bdf -o digitalnumbers-68.c -n digitalnumbers_68_tn -d ../bdf/helvB18.bdf 

[image]

c-code:
[digitalnumbers-68.txt](https://github.com/olikraus/u8g2/files/966096/digitalnumbers-68.txt)

olikraus, 2017-04-29:
hmmm i just realized, that you placed some more glyphs into this bdf file.

cmd:

./bdfconv -v -f 1 -m  '32,48-57,8000-9000' digitalnumbers-68.bdf -o digitalnumbers-68.c -n digitalnumbers_68_ta -d ../bdf/helvB18.bdf 

[image]

[digitalnumbers_68_ta.txt](https://github.com/olikraus/u8g2/files/966099/digitalnumbers_68_ta.txt)

koenjer, 2017-04-30:
Hi Oli,

Thanks for the quick reply and help! Got it working :)

Another question, it is possible to change the font position.
There are options for bottom/top/center but is there also an option to have
a start position on the right side of the text/numbers?

I'm making a weighing system and a number is encounting so you would like
to have a fixed point on the right side of the numbers.

Thanks in advance for your answer.

Best regards,

Gijs

olikraus, 2017-05-01:
There is a function to calculate the width of the string. Calculate this width first, then substract that value from the desired x position to get the position for the draw string procedure.

koenjer, 2017-05-01:
Hi Oli,
I'm new in the arduino /C programming world. So i'm sorry for my novice questions.
I'm strugling with the getstrwidth command.

I need to know the width of a long integer. (weight)
But I get a error "invalid conversion" 

code used:
    const char *charValue = weight;
    u8g2.setCursor(144 - u8g2.getStrWidth(charValue), 44);

What am I doing wrong?

olikraus, 2017-05-02:
Maybe post some code. Such a problem is mostly caused by reading the time value within the picture loop.

koenjer, 2017-05-02:
I didn't know how to add a font to the existing font libary so I replaced a existing font with the font code you supplied in a few postings above.

This error comes back from the compiler:
C:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\clib\u8g2_fonts.c:120258:3: warning: initializer-string for array of chars is too long

   "\23\0\5\5\6\6\4\7\7<,\0\0+\0+\0\0\0\0\0\1O \6\0\200A(\60&\333"

   ^

olikraus, 2017-05-02:
The warning can be ignored, but maybe something bad happened during copy & paste.
Still i think this problem is caused by your draw procedures. Maybe you can post you drawing code here.

koenjer, 2017-05-03:
it is not a copy/paste issue. But the problem only exist with this new font. (other fonts do not have the issue)

here is a copy of my code:

void loop(void) {
  static time_t tLast;
  time_t t;
  tmElements_t tm;
  
  u8g2.firstPage();

  do {
    u8g2.setFontMode(1);  /* activate transparent font mode */
    u8g2.setDrawColor(1); /* color 1 for the box */
    u8g2.drawBox(0,0,240,118); 
    u8g2.setDrawColor(0);
    u8g2.drawHLine(0,60,240);
    u8g2.drawHLine(0,61,240); 
    u8g2.drawHLine(0,75,240);
    u8g2.drawHLine(0,76,240); 
    u8g2.drawTriangle(211,49, 224,57, 211,57);
    u8g2.drawTriangle(235,57, 221,57, 235,49);
    u8g2.setFont(u8g2_font_courB18_tf );
    u8g2.drawStr(178,44,"kg");
    u8g2.setDrawColor(1);
    u8g2.setFont(u8g2_font_6x10_mf);
    u8g2.drawStr(10,127,"MENU");
    u8g2.setDrawColor(0); 
    u8g2.setFont(u8g2_font_px437wyse700b_mn);
    char charValue[5]; 
    String str; 
    str=String(waarde); 
    str.toCharArray(charValue,5); 
    u8g2.setCursor(174 - u8g2.getStrWidth(charValue), 44);  

    u8g2.print(waarde);

    RTC.get(); 
    RTC.read(tm); 
    u8g2.setFont(u8g2_font_6x12_mf);
    u8g2.setCursor(30, 72);
    u8g2.print("Tijd ");
    print2digits(tm.Hour);
    u8g2.print(":");
    print2digits(tm.Minute);
    u8g2.print(":");
    print2digits(tm.Second);
    u8g2.print("  Datum ");
    print2digits(tm.Day);
    u8g2.print("/");
    print2digits(tm.Month);
    u8g2.print("/");
    u8g2.print(tmYearToCalendar(tm.Year));

  } while ( u8g2.nextPage() );
  
  delay(100);

olikraus, 2017-05-03:
Quoting myself:

>  Such a problem is mostly caused by reading the time value within the picture loop.

after seeing your code:

"The problem is  caused by reading the time value within the picture loop."
Move
```
RTC.get(); 
RTC.read(tm); 
```
before the 
`u8g2.firstPage();`
command.

koenjer, 2017-05-03:
sadly that did not fix the issue. 
Even with this bare minimum code I get a corrupt screen when using the number "9".
So to my opinion there is something wrong with the rendered font.
Could you try a new or different way to render the font?

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setDrawColor(1); /* color 1 for the box */
    u8g2.setFont(u8g2_font_px437wyse700b_mn);
    u8g2.setCursor(144, 44);  
    u8g2.print("3900");
  } while ( u8g2.nextPage() );
}

olikraus, 2017-05-03:
Can post a screenshot and the complete code for testing this (including constructor)?

koenjer, 2017-05-03:
Yes no problem.

Code:

#include <Arduino.h>
#include <U8g2lib.h>

U8G2_T6963_240X128_1_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable=*/ 17, /*cs=*/ 14, /*dc=*/ 15, /*reset=*/ 16); // Connect RD with +5V, FS0 and FS1 with GND

void setup(void) {
pinMode(18, OUTPUT); 
digitalWrite(18, 1);  
u8g2.begin(/*Select=*/ 53, /*Right/Next=*/ 0, /*Left/Prev=*/ 0, /*Up=*/ 51, /*Down=*/ 49, /*Home/Cancel=*/ 46);  
}

void loop(void) {
  u8g2.firstPage();

  do {

    u8g2.setDrawColor(1); /* color 1 for the box */
    u8g2.setFont(u8g2_font_px437wyse700b_mn);
    u8g2.setCursor(100, 50);  
    u8g2.print("3900");
  } while ( u8g2.nextPage() );
}

[image]

koenjer, 2017-05-03:
I must admit that I do not know how to "add" the font you created so I did a copy paste of the code in a existing font in the file u8g2_fonts.c
Not sure if that creates a problem?

olikraus, 2017-05-03:
can you test this font?

```
const uint8_t digitalnumbers_68_tn[347] U8G2_FONT_SECTION("digitalnumbers_68_tn") = 
  "\13\0\5\6\5\6\4\2\7\36+\0\0+\0+\0\0\0\0\0\1B \6\0\300\241\0\60%{"
  "]\303\342\62\62\341\7.Q\371\377b!\14\346`\4j`\14\346 \24\242\362\377\305\37\270\204\314\270"
  "\10\0\61\27E\365\245\42\60\6\341\7\66a\60\2c\20~\300\22\6# \0\62$ze\303\302\62"
  "\60A\23\62\2S\213\377o\134\4&`\211\313 X\342\377Y\214\310\4M\300\14\213\0\0\63\37y"
  "m\303\242\62.!\23\60\342B\213\377O\134\302\325\270\242\305\377'.!\26.\243\42\0\64\35[\335"
  "\303\42P\3c\60\7\241\20\225\377\277\250\13\231\221e\213\377\277;\310\1\1\0\65\42{]\303\342\62"
  "\62\241\26\62\242`\213\377g\63B\23\66A#\62\266\370\377D&\324Bf\134\4\0\66 {]\303"
  "\342\62\62\241\26\62\242`\213\377w\61\4\63\62A\27\225\377\27\x7f\340\22\62\343\42\0\67\36Z\345\303"
  "\302\62\60A\23\62\2S\213\377\317\16p\0\17l\241\26\377\377\354\0\7\4\0\70 {]\303\342\62"
  "\62\341\7.Q\371\x7f\21\24\62\343\62\62\1\24\225\377\27\x7f\340\22\62\343\42\0\71\36{]\303\342\62"
  "\62\341\7.Q\371\x7fQ\27\62#\313\26\377\237\310\204Z\310\214\213\0\0\0\0\0";
```

olikraus, 2017-05-03:
```

const uint8_t digitalnumbers_68_tn[347] U8G2_FONT_SECTION("digitalnumbers_68_tn") = 
  "\13\0\5\6\5\6\4\2\7\36+\0\0+\0+\0\0\0\0\0\1B \6\0\300\241\0\60%{"
  "]\303\342\62\62\341\7.Q\371\377b!\14\346`\4j`\14\346 \24\242\362\377\305\37\270\204\314\270"
  "\10\0\61\27E\365\245\42\60\6\341\7\66a\60\2c\20~\300\22\6# \0\62$ze\303\302\62"
  "\60A\23\62\2S\213\377o\134\4&`\211\313 X\342\377Y\214\310\4M\300\14\213\0\0\63\37y"
  "m\303\242\62.!\23\60\342B\213\377O\134\302\325\270\242\305\377'.!\26.\243\42\0\64\35[\335"
  "\303\42P\3c\60\7\241\20\225\377\277\250\13\231\221e\213\377\277;\310\1\1\0\65\42{]\303\342\62"
  "\62\241\26\62\242`\213\377g\63B\23\66A#\62\266\370\377D&\324Bf\134\4\0\66 {]\303"
  "\342\62\62\241\26\62\242`\213\377w\61\4\63\62A\27\225\377\27\x7f\340\22\62\343\42\0\67\36Z\345\303"
  "\302\62\60A\23\62\2S\213\377\317\16p\0\17l\241\26\377\377\354\0\7\4\0\70 {]\303\342\62"
  "\62\341\7.Q\371\x7f\21\24\62\343\62\62\1\24\225\377\27\x7f\340\22\62\343\42\0\71\36{]\303\342\62"
  "\62\341\7.Q\371\x7fQ\27\62#\313\26\377\237\310\204Z\310\214\213\0\0\0\0\0";

void loop(void) {
u8g2.firstPage();
do {
u8g2.setDrawColor(1); /* color 1 for the box */
u8g2.setFont(digitalnumbers_68_tn);
u8g2.setCursor(144, 44);
u8g2.print("3900");
} while ( u8g2.nextPage() );
}

```

koenjer, 2017-05-04:
Yes, copying the font directly into the program works!

closed 2017-05-14 by olikraus

marcelohpn, 2017-12-14:
[digital7.zip](https://github.com/olikraus/u8g2/files/1557691/digital7.zip)
Hi there! I´m having some throuble to find a way to convert ttf to bdf. I´m using win7. 

Can you please help? I only need ascii 45 to 58.

Font size 28 and 32.

I usually use adafruit library, but I´m working with a oled that adafruit don´t support. 

Thank you for doing a great job.

olikraus, 2017-12-22:
To my knowledge you can use fontforge for ttf to bdf conversion. But i do not know how to use this. I personally use otf2bdf (linux) for this purpose.

g2tec, 2018-05-10:
Hello, Oliver. I've added the "digitalnumbers-68" font to my project but the number 1 does not have the correct spacing for a 7-segment display simulation. I modified the BDF file on FontForge but I don't know how to convert to .c using the bdfconv exe. 

Attached is the BDF file of the corrected font. The extension is changed to txt so github allows the upload.

[digitalnumbers2-68.bdf.txt](https://github.com/olikraus/u8g2/files/1993109/digitalnumbers2-68.bdf.txt)

Thank you.

olikraus, 2018-05-11:
What is the problem? A windows version of the bdfconv.exe is here: https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/bdfconv_2_22.exe and the comandline call for this font is mentioned somewhere above.

g2tec, 2018-05-11:
Running the command"cmd /K C:\bdfconv_2_22.exe -v -f 1 -m "32,48-57" -n digitalnumbers2_68_tn -o digitalnumbers2-68.c digitalnumbers2-68.bdf -v" returns "Can not open bdf file 'digitalnumbers2-68.bdf'

olikraus, 2018-05-11:
Did you copy the bdf file to "c:"? Otherwise you need to provide the full path for the bdf file.
I have created a .c file with the latest version of bdfconv:

```
/*
  Fontname: -FontForge-DSDigital-Book-I-Normal--68-650-75-75-P-287-ISO10646-1
  Copyright: Font Typeface: DS-Digital. Created by Dusit Supasawat , DS-Font 1998. All Rights Reserved
  Glyphs: 11/13
  BBX Build Mode: 0
*/
const uint8_t digitalnumbers2_68_tn[355] U8G2_FONT_SECTION("digitalnumbers2_68_tn") = 
  "\13\0\5\6\5\6\6\2\7\36+\0\0*\0*\0\0\0\0\0\1F \6\0\0\207\2\60%{"
  "\35\15\213\313\310\204\37\270D\345\377\213\205\60\230\203\21\250\201\61\230\203P\210\312\377\27\177\340\22\62\343"
  "\42\0\61\30E\315\15\213\300\30\204\37\330\204\301\10\214A\370\1K\30\214\200\0\0\62$z%\15\13"
  "\313\300\4M\310\10L-\376\277q\21\230\200%.\203`\211\377g\61\42\23\64\1\63,\2\0\63 "
  "y-\15\213\312\270\204L\300\210\13-\376\77q\11W\343\212\26\377\237\270\204X\270\214\212\0\0\64\35"
  "[\35\17\213@\15\214\301\34\204BT\376\377\242.dF\226-\376\377\356 \7\4\0\65\42{\35\15"
  "\213\313\310\204Z\310\210\202-\376\237\315\10M\330\4\215\310\330\342\377\23\231P\13\231q\21\0\66!{"
  "\35\15\213\313\310\204Z\310\210\202-\376\337\305\20\314\310\4]T\376_\374\201K\310\214\213\0\0\67\36"
  "Z%\17\13\313\300\4M\310\10L-\376\77;\300\1<\260\205Z\374\377\263\3\34\20\0\70!{\35"
  "\15\213\313\310\204\37\270D\345\377EP\310\214\313\310\4PT\376_\374\201K\310\214\213\0\0\71\36{"
  "\35\15\213\313\310\204\37\270D\345\377E]\310\214,[\374\177\42\23j!\63.\2\0\0\0\0\4\377"
  "\377\0";
```

But the font format is changed, so in order to use this, you need to download & install latest beta of u8g2: https://github.com/olikraus/U8g2_Arduino/archive/master.zip

g2tec, 2018-05-11:
All the files were in C:\. I think that the stupid Windows  UAC control was not allowing the cmd to read/modify it.

Thank you for converting the bdf file. My project can now live on.

djairjr, 2021-03-06:
I've made that one, converting DS-Digital.

/*
  Fontname: u8g2_font_clock_font
  Copyright: 
  Glyphs: 59/59
  BBX Build Mode: 0
*/
const uint8_t .\u8g2_font_clock_font_tn[3532] U8G2_FONT_SECTION(".\u8g2_font_clock_font_tn") = 
  ";\0\5\4\5\6\4\7\7#\62\0\0+\0+\11\6h\0\0\15\263 \6\0@`\23!!("
  "m\341\363\4#\312 dT\242\214\372\21R\302\15f\20\62\352o\220\62\214\70a\227Q\67\0\42\17"
  ",a\61\24\24\241\237H\203\10Q\204\0#g\273c%\70\211#\36)\244\25bX!\245\25RZ"
  "!\245\225B\322\20#\220\60\304 \225\10\301\210K\240`H!f\20!\304\20B\220D\12i\245\24"
  "VHi\205\224\64\304\20BL\244\210 \210\10\242\210\63\314\70\341\14\63\216(\42\10\42\202(i\230"
  "\341\12)\314\220\302J)\254\20\303J!n\34\1\3\12\12\0$>\370\335\341\266\5;h\231\65Y"
  "\242K\17\204\363\300\70\17\204\42\204+d\326o\22:\2K!\270\364@Hn\61\21\354\240d\326o"
  "\222\342\204(\17\204\363\300\70\17\204\344^\231ut\64\0%v\376lcy\17\134\16i\306\230e\14"
  "R\5\31e\212\20D\204c\12\61\303\224R\21S\252aL)e\224SJ\21\346\224B\204IB\20"
  "!Fi\306\24\206\314p\346\10H>$F.\205<C\214+\4\61S\14\63C\10\42D*\203\224"
  "r\214(\245\30\63J)\246\220RJ\61\244\224B\214\31\206\224\202\202 B\24\243\14*\353\34\322J"
  "\32N,\301C\17\31\0&UZ\334d\27\251\307\232cm\5\261\326\260\245\326Z\70\205\215RX!"
  "e\31RV)$\254@B \42\264`F\10/\230\322\310\30a\254\21D\30\343\215B\32)e\25"
  "bV!\205\25RX!\205\25RV)e\225R\30\61B(\61\24\23A\271\306\236Z\0'\13%"
  "a\261\22| \11!\0((N\334d\24\231s\314\71\246\10A\12Q%\25dPIu\211\254\360"
  "\201\12l\250\222\352\227\310\22b\240\20\12\272\11\0)(O\324dt\34\204!\243\306\10m\254\242j"
  "\251\250j\21\66\230\370@\217U\222IE\325S\344\14!NI\347\134\10\0*>\66[\251v\5:"
  "d\211\265TF\21&\30A\304\11G\214p\304\21\42\240qB\20\307\224a\274\21\306\234\20\204!'"
  "\10q\5\42\214\70\201\210\23\312\250R!\201\25h\42\221\203\6\6\0+$\324\342g\66\11\71`y"
  "up(#B\60\3\11\24\36(\1\11\64\14\61,\304\1\313\253\203C\212\4\0,\15\6Q\242\62"
  "\250P'\6\11\3\0-\24\264`lV<\20\306\3#<P\302\3c<\20\4\0.\13\246\320\244"
  "\62\224\240B\25\0/.\230T\244\326\212Y\244\211F\232hI\23-i\242\221e\226I\350\370P\11"
  "\264H\23\215\64\321&M\264\244\221e\22J\350\250\342\6\13\0\60dX\334\344\226<@H\10\17\210"
  "\20\206\10N\10A\2\23C\20\241\4\21\205\25QX\21e\225QV\31e\225QV\31e\225QV"
  "\31\204\225\61\34!\1\212\17\241P\304\33\203\64\62\312*\243\254\62\312*\243\254\62\312*\242\260\42\12"
  "+\242\260\42\210P\241\210!X\30C\204\27\304\10\341\1\21\2y\200\20\0\61\35\347\353\345\323\204\42"
  "\310\30D\224PD\375\6!\201\206\62\6\21\365\67F\21&\14\0\62\71X\334\344\226<@\314\3\42"
  "\4\343\204\70L\14\244\4\231\225,\263\276\303\2\61.\10\363@Hn\61$*\241e\326\311\62\253I"
  "\204BC\260#\302\63!< \314\3\204\0\63\67W\344\344v<@\312\3\42\204\342\204\60L\214\243"
  "\4\221U,\262>\303\2).\210\362@@N\61\21\352\230D\326\337Q\241\30\26FyA\220\7D"
  "\10\343\1B\0\64\65\366\343\345\266\16Y\340q\211\251\265V\11e\225PV\11e\225PV\11e\225"
  "@X\11#\260@D\10.\10\362@\70.\61\21\350\220$\326\237$S\324\60\0\65\67V\344\344v"
  "<@D\10\17\210!\202#$\260B\204\62%\326/\22\71\2;!\270\363@\70.\61\21\350\220$"
  "\326\237Q\241\24\26\6yA\214\7D\10\342\1\62\0\66QW\334\344\226<@F\10\17\10\42\202+"
  "$\60C\204:E\326O\222\71\2C!\70\364@@N\61\21\210xC\220FDYE\224UDY"
  "E\224UDY%\24VBa%\24V\2\21*\224\60\4\13C\210\360\202\20!< B\30\17\220"
  "\1\0\67'\24\364\345\26<@\304\3\42\4\341\204\30L\14\242\4y\225+\257>H\242\370`\7\71"
  " y\365\7I\24\63\14\0\70jX\334\344\226<@H\10\17\210\20\206\10N\10A\2\23C\20\241"
  "\4\21\205\25QX\21e\225QV\31e\225QV\31e\225QV\31\204\225\61\2\13\204\204\340\202\60"
  "\17\204\344\26\23\241\210\67\6id\224UFYe\224UFYe\224UDaE\24VDaE\20"
  "\241B\21C\260\60\206\10/\210\21\302\3\42\4\362\0!\0\71QW\344\344v<@F\10\17\210\20"
  "\204\10N\210@\2\23#\20\241\4\11\205\225PX\11e\25QV\21e\25QV\21e\25QV\21"
  "\204\25\61\2\13d\204\340\202(\17\4\344\24\23\241\216Id\375\35\25\212aa\224\27\4y@\204\60"
  "\36 \4\0:\20\250\333$s\324\211\362\341|\31\65Q\6\0;\24H\134\42s\324\211\362\341\235\62"
  "\212(\243\216\14\23\12\0<\31\17[\347t\21U\222A\347\334\337=\351\226\214\262\322IF\225E\4"
  "\0=\42\265Yjv< \306\3$<P\304\3\203<\20>d\36\10\343\1\22\36\60\341\1\62\36"
  "\20\2\0>\31\17[\347T\220UTQV\272\222QV\272\354\71\367C&\25E\26\0?\63\71U"
  "\341\226<\20\324\3A\275\325\332\22\302\222Zf\241\365!\26\310qA\234\7\202r\214%aI-\264"
  "n\26Z\243\244\16\33>\314\24Zf\241\25\5@p^\334d\30=P\322\3E\4\363@\21\342<"
  "\60\204(B<\20\304 D\222Qb\31%\226Q$\31\5%!H\71\353\24\323L\61\354\20\42\204"
  "\22\202\14B\330(\241\24EVIe\225TVI\205\210RR\31\204\20UF!B\244@F\71J"
  "\214Q\14\23b\224\303D\30\5\61Rr\321d\13\361@\61A<p\314\3\312<\200\320\3\206\0A"
  "dX\334\344\226<@H\10\17\210\20\206\10N\10A\2\23C\20\241\4\21\205\25QX\21e\225Q"
  "V\31e\225QV\31e\225QV\31\204\225\61\2\13\204\204\340\202\60\17\204\344\26\23\241\210\67\6i"
  "d\224UFYe\224UFYe\224UDaE\24VDaE\24V\4qc\14(\206\220a\204"
  "\31\6\0B[X\334\344\226<\24\302;\42\70C\2;D(!Had\24VDYe\224UF"
  "Ye\224UFYe\224U\6ae\214\300\2!!\270 \314\3!\271\305D(\342\215A\32\31e"
  "\225QV\31e\225QV\31e\25QX\21\205\25Q\30\31$,!\310\10\355\210\340\222\227\0C."
  "X\334\344\26=\20\316CO\65$\304Bd\326o\22:l\370\260 *\241e\326\311\62\13%t\210"
  "\225\202h\351\201\220\36\10\351\201\60\0DUX\334\344\226<\24\302;\42\70C\2;D(!Ha"
  "d\24VDYe\224UFYe\224UFYe\224U\6ae\14GH\200\342C(\24\361\306 "
[... 25 lines cut ...]
  "ad\24VDYe\224UFYe\224UFYe\224U\6ae\214\300\2!!\270 \314\3!"
  "\271\305\220\250\204\226Y'\313\254MBG\25\66\134\0QWW\334\344\26)\306\224K\15\5\261\204 "
[... cut ...]
  "W\334\244v\10(\306pC\24FDaD\30U\204If\230sH\71\246X\307\14\203\214\70\251\10"
  "\263\210(\214\10\342\206 n\210\361\201\37b\70\42\206#\202\260\42\310\62\242$\63J\62\303\34S,"
  "c\310\71f\230T\204Q&\24FDaD\20\67\304xC\0Y=V\344\344\266\206\20\244\10\2\36"
  "\227\230n\225PV\11e\225PV\11e\225PV\11\204\225\60\2\13D\204\340\202 \17\204\343\22\23"
  "\201\16Ib\375\42)L\210\342\316\3\341\270\304\20\0Z\67X\334\344v<@\314\3\304<@\314\3"
  "\2\275\17\30\231\225\64\321H\23-id\231\202\206:(\231EZ\321\222F\226I>`.=\20\316"
  "\3\343<\60\316\3\203\0\0\0\0";

v1993, 2021-03-06:
@djairjr Please enclose your code into ``` on each side, it's unusable right now.

olikraus, 2021-03-07:
code quotes added...

sjorsvr, 2022-03-28:
Hi I would like to add a new font to u8g2 but do not know how to do it. I want to have the 8544 char font as defined as 
f.i. in the pdf8544 arduino library 👍 
// The 7-bit ASCII character set...
const PROGMEM unsigned char charset[][5] = {
  { 0x00, 0x00, 0x00, 0x00, 0x00 },  // 20 space
  { 0x00, 0x00, 0x5f, 0x00, 0x00 },  // 21 !
  { 0x00, 0x07, 0x00, 0x07, 0x00 },  // 22 "
  { 0x14, 0x7f, 0x14, 0x7f, 0x14 },  // 23 #
  { 0x24, 0x2a, 0x7f, 0x2a, 0x12 },  // 24 $
  ....
available in u8g2
How can I do this ?
regards
Sjors

olikraus, 2022-03-28:
> How can I do this ?

You need to create a .bdf file. fony.exe usually is a good windows tool to do this.
Then convert the .bdf file into the internal data format (using bdfconv), which in turn can be used as argument to setFont command.

U8g2 faq contains some description for this.

sjorsvr, 2022-03-28:
Thanks, I have a look at it
Regards
Sjors