olikraus/u8g2 #254: Support for OLED 64x32

sympthom9, 2017-04-30:
Hi,

Thanks for a great library and great job!

Are the support coming for the 64x32 OLED display in the near future?

Cheers,
Martin

olikraus, 2017-04-30:
Any datasheet available?
Can you do testing for me?

sympthom9, 2017-04-30:
Should be it: http://www.buydisplay.com/download/manual/ER-OLED0.49-1_Series_Datasheet.pdf

Sure, I will give it a go.

sympthom9, 2017-04-30:
http://shop-on-line.tvielectronics.com/download/TOL049G-6432W_DataSheet.pdf

olikraus, 2017-05-01:
Can you test the U8G2_SSD1306_64X48_ER device? what will you see? Can you send me a picture of the hello world example?

sympthom9, 2017-05-01:
I've ordered a few devices. As soon I get them by mail I will be able to do some testing.

Here is a port already for the 64x32: https://github.com/WaiakeaRobotics/Adafruit_SSD1306

olikraus, 2017-05-01:
Where did you order these devices? Maybe i can also order one.

sympthom9, 2017-05-01:
Here is one that I found: http://www.ebay.com/itm/332066813244?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

olikraus, 2017-05-01:
Hmm this seller does not ship to my country.
I will order this:
http://www.ebay.com/itm/White-0-49-inch-OLED-Display-Module-64x32-0-49-I2C-IIC-for-Arduino-AVR-STM32/371842373740

olikraus, 2017-05-21:
OLED did not yet arrive at my lab. moving this to next milestone.

sympthom9, 2017-05-21:
Yep, same situation here, still waiting on mine to arrive.. China shoud be a whole lot closer..

sympthom9, 2017-05-31:
I finally got the OLED.. I'm doing the coding and am starting from the 64x48 as a boilerplate. Any pro tips from you as I progress?

olikraus, 2017-05-31:
I guess I need to create a device for this first. Then we both can modify this, until the display works correctly.

olikraus, 2017-05-31:
I created a new 64x32 device as a copy of the 64x48.
Download the new u8g2 version here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
You need to manually add this zip file via the Arduino IDE "Add Zip Library" menu.

All examples now contain a 64x32 constructor.
For example the u8x8 flipmode has this:
U8X8_SSD1306_64X32_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);  

Of course all other types of constructors also  exist (HW_SPI, SW_SPI, etc)

there is also a new file: u8x8_d_ssd1306_64x32.c The content of this file needs to be modfied for your display. Maybe you can send me the output of HelloWorld.ino so that we can update this file together.

sympthom9, 2017-05-31:
Oh, I've just finished the code myself and was going for a pull request. :) Well nicely done on your end. I will undo my code and will start test 2.16.0 instead.

olikraus, 2017-08-04:
Any further feedback on this?
@sympthom9 does this work for you?

olikraus, 2017-08-12:
closing this, lets see whether this works...

k1hop, 2017-09-18:
hi olikraus,  Not sure if it's okay for me to leave another comment after you closed this issue, but if not, just tell me where to post it.

I recently purchased a 64x32 OLED display from ICStation.com (http://www.icstation.com/blue-049in-oled-display-module-64x32-screen-arduino-stm32-p-10787.html) and was having some troubles while using u8x8 library to drive it with an Arduino.

I was able to determine the I2C address and talk to it and get "stuff" to appear on the screen.  However, first I noticed that the u8x8.clear() function only cleared about the top half of the display and left RAM random garbage on the bottom half.  I read that one might have to alter the u8x8_d_ssd1306_64x32.c file for each particular display, so I started playing with that.

First, I changed the DISPLAY OFFSET value and was able to move things around a bit to where it looked pretty normal.  It appeared that for some reason the entire display was "shifted up two lines" (for text) so the first two lines were off the display entirely, and the last two lines were in the top half.  But below these lines in the bottom half was still the random garbage which was never being cleared.

On a lark (not really knowing why or what I was doing) I decided to put the multiplex value back to 0x2f instead of the 0x1f the comment said it had been changed to.  I reset my DISPLAY OFFSET to 0x000 and voila, everything looked perfect... and addressing was correct... AND the garbage was now being cleared by the u8x8.begin() command!!

A bonus is that the display appeared brighter to my eyes... though maybe that was  psychological due to the garbage disappearing and everything looking cleaner.  But perhaps with a different multiplex value it was in fact brighter???

Anyway, thought I'd pass on these notes re this particular display.  I don't know if it's the same as the ones you guys played with months earlier or not.  I have NO specs on the display itself, but it's working great with this change!

If I had not found this post, I'd probably have thrown my hands up in frustration by now! :)

Russ K1HOP

olikraus, 2017-09-19:
Sounds great. Can you share the modified code by pasting it here?

k1hop, 2017-09-19:
No sense posting the entire program for an ultimate change of ONE CHARACTER! :)
In the program u8x8_d_ssd1306_64x32.c this is the one and only change that ultimately did the trick:

  U8X8_CA(0x0a8, 0x1f),		/* multiplex ratio: changed from 0x1f to 0x2f */

As you can see from YOUR original comment which I did not change, it had been changed from 0x1f to 0x2f.  I simply changed it back to 0x1f again.

Russ

olikraus, 2017-09-19:
Ah, ok, great. Now I understand your modification. :)

olikraus, 2017-09-19:
Will continued here: #361

closed 2017-09-19 by olikraus