olikraus/u8g2 #255: RAM drain T6963

fire1, 2017-05-01:
I'm using 
`lcd->firstPage();
do{
menus();
} while (lcd->nextPage());`
without any delay... (just as information)

My program starts with 6057 bits of free RAM but after draw is runned free RAM is 5897

This is my dump from serial: (first draw initialization)
```
 Start (RAM): 6057
 After draw (RAM): 5897
 End (RAM): 5897
```
For the next draw initialization usage is: (second draw initialization)
```
Start (RAM): 5897
 After draw(RAM): 5801
 End  (RAM): 5801
```
And like so, to end of the free RAM ....

When i remove u8g2  drawing  from sketch everything is fine, also i try with clearBuffer() method placed before/after the drawling call.
I try all constructors for 8080's T6963... same issue (preferred "page first/next" works faster ...) 
What can i do to find where is causing this memory leak...

P.S.
Great thanks to the great library!!!

olikraus, 2017-05-01:
U8g2 does not allocate memory. Something else must cause this effect.

fire1, 2017-05-01:
Sorry my bad! 
I found a bug after 1 month of searching ...

closed 2017-05-01 by fire1