olikraus/u8g2 #258: Hardware SPI collides with SPI-ISR

gpx1, 2017-05-02:
Hi Oli,
Great library by the way! ... But, I got a problem (system crash) during the integration into our environment where already interrupt routines use HW SPI. I actually didn't find the part in the code to check where the access to the SPI registers occure. Is it possible that during SPI page transfer interrupts aren't disabled?
Looking forward to getting your reply.
Regard, Gerhard

olikraus, 2017-05-03:
Arduino IDE? Which constructor?
U8g2 itself does not access any registers. Only the Arduino SPI and digitalWrite procedures are called.

gpx1, 2017-05-03:
Oh, sorry. That was a quick response!

Arduino IDE: Genuino 1.6.12 (don't want to switch IDE during a project)
Library: U8g2
Constructor: U8G2_UC1701_EA_DOGS102_1_4W_HW_SPI(rotation, cs,  a0, reset)

I also set:
#define U8X8_HAVE_HW_SPI

Just in case:
The SPI transfer method (Arduino) should be encapsulated by disable/enable Interrupt commands like in that example. 

Thanks a lot and best regards
gpx

// example for SPI transfer method within an interrupthandler ===============
void RFM69::writeReg(uint8_t addr, uint8_t value)
{
  select();  // disable interrupts
  SPI.transfer(addr | 0x80);
  SPI.transfer(value);
  unselect();   // enable interrupts
}

void RFM69::select() {
  noInterrupts();
#if defined (SPCR) && defined (SPSR)
  // save current SPI settings
  m_SPCR = SPCR;
  m_SPSR = SPSR;
#endif
  // set RFM69 SPI settings
  SPI.setDataMode(SPI_MODE0);
  SPI.setBitOrder(MSBFIRST);
  SPI.setClockDivider(SPI_CLOCK_DIV4); // decided to slow down from DIV2 after SPI stalling in some instances, especially visible on mega1284p when RFM69 and FLASH chip both present
  digitalWrite(m_slaveSelectPin, LOW);
}

void RFM69::unselect() {
  digitalWrite(m_slaveSelectPin, HIGH);
  // restore SPI settings to what they were before talking to RFM69
#if defined (SPCR) && defined (SPSR)
  SPCR = m_SPCR;
  SPSR = m_SPSR;
#endif
  // Only reenable interrupts if we're not being called from the ISR
  if (!_inISR) interrupts();
}

olikraus, 2017-05-03:
> The SPI transfer method (Arduino) should be encapsulated by disable/enable Interrupt commands 

If this is required for your board, then this should be done inside of the SPI transfer method.
It does not make sense, that each application of the SPI transfer methods adds such code separatly.

What i also do not understand is, why you need to set U8X8_HAVE_HW_SPI.
It should be detected automatically for the Arduino Uno Board.

gpx1, 2017-05-03:
Thanks, I didn't know that this constant will be set automatically. I'll remove that.

The SPI transfer method is part of the Arduino SPI library. To docter an official general library is one thing I realy want to avoid. Besides that it woudn't solve my problem.
The core of my problem is that the Ug82-HW-SPI-transfer gets interrupted by my interrupt service routine in a critical SPI phase. Usually each application using SPI makes sure not to get interrupted in a critical SPI operation by another SPI application.

For me solving the problem means doctoring your library which is not desirable.

Thanks anyway,
Gerhard

olikraus, 2017-05-03:
You could surround the u8g2.nextPage() or u8g2.sendBuffer() procedure (whatever is used by you) with the disable/enable irq command.

olikraus, 2017-05-03:
I just went through the AVR code of SPI library (arduino-1.8.0/hardware/arduino/avr/libraries/SPI/src). There is a lot of interrupt protection already inside of the Arduino SPI lib. Did you register your interrupt procedure for the SPI lib?

https://github.com/arduino/Arduino/pull/2381

gpx1, 2017-05-03:
Thanks, I registered this instance with the usingInterrupt() method of the SPI-class if that is what you mean, but I'll doublecheck that. I already have a SPI-flash and a SPI-radio chip where the radio is interrupt driven. 
Anyway, I still couldn't find in the U8g2 code the actual call to the SPI methods. Is that in some precompiled lib-file? If not I'd be greatful if you point me to where I should look.
I already enclosed both, the firstpage and the nextPage call with interrupt blocks. Unfortunately I got some sideeffects with missing telegrams. But again I need to have a closer look to that.
I'll get back to you as soon I know more.
Thanks gpx

olikraus, 2017-05-14:
The calls are done in U8x8lib.cpp

Any news on this, can i close this issue?

gpx1, 2017-05-14:
Thanks for the input. I checked the possibility to place the disable/enable interrupt further in the u2g2lib. There is just a small gain in cycle time as a trade off to stay in line with the main branche of u8g2. So I decided not to disrupt the lib and keep it as it is. I can live with this compromise.
Anyway, I really love this library. The graphics functionallity is outstanding.
Thank again and best regrds, gpx

closed 2017-05-14 by gpx1

olikraus, 2017-05-14:
:)