olikraus/u8g2 #261: SAMD21 support?

anhtuanmanga, 2017-05-04:
I bought the samd21 broad
http://robotdyn.com/catalog/boards/samd21_m0_mini/
Do your lib work on this broad
Sorry for bad english, this is not may native language

olikraus, 2017-05-04:
If this board works with the Arduino IDE, then probably yes.
U8g2 just uses Arduino function and should be compatible with all Arduino Boards.

anhtuanmanga, 2017-05-05:
I succes to upload code to broad but not thing show on oled.
Did you test with arduino zero before?

olikraus, 2017-05-05:
I did tests with the Feather M0 Board.
In most cases this is a wiring problem.

anhtuanmanga, 2017-05-05:
Check every thing but still not work.
May be problem is my broad.
Thanks for your help

olikraus, 2017-05-05:
Feel free to send more information for cross check: pic of setup, code, ...

anhtuanmanga, 2017-05-07:
here is my code

#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SSD1306_64X48_ER_1_SW_I2C u8g2(U8G2_R0, 7, 8);
void setup(void) {

  u8g2.begin();  
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
}
here is my setup
[image]

olikraus, 2017-05-07:
I might be wrong but it looks like d9 is connected and not d8

anhtuanmanga, 2017-05-07:
no it is d8

olikraus, 2017-05-07:
Then, I also don't know. All looks correct.

anhtuanmanga, 2017-05-08:
me too, may be because of my broad
thanks for your support

olikraus, 2017-05-14:
not sure how to support further... closing this...

closed 2017-05-14 by olikraus

anhtuanmanga, 2017-05-18:
i  change to platformio ide and everything work.
don't know why.

olikraus, 2017-05-18:
Somethimes there are breadboard issues...