olikraus/u8g2 #262: High RAM usage vs. u8glib

SukkoPera, 2017-05-07:
OK, perhaps this is normal, but I just thought I'd ask to make sure: I have ported an Arduino sketch of mine from u8glib to u8g2 (See https://github.com/SukkoPera/MyTrackr/commit/a160ea4094dde1befe84afc6135a34d0929cdbe7). The updated version uses 354 bytes more RAM, which means the sketch won't even work as expected, because 97% RAM gets used. For reference:

Before:

> Lo sketch usa 25828 byte (80%) dello spazio disponibile per i programmi. Il massimo è 32256 byte.
> Le variabili globali usano 1645 byte (80%) di memoria dinamica, lasciando altri 403 byte liberi per le variabili locali. Il massimo è 2048 byte.

After:

> Lo sketch usa 27162 byte (84%) dello spazio disponibile per i programmi. Il massimo è 32256 byte.
> Le variabili globali usano 1999 byte (97%) di memoria dinamica, lasciando altri 49 byte liberi per le variabili locali. Il massimo è 2048 byte.

On a side note, u8g.getStrWidth() was also working with Flash strings, how do I do that now?

olikraus, 2017-05-07:
U8g2 will use standard Arduino libs. These libs require more ram :(

SukkoPera, 2017-05-07:
Too bad :(. What about the question in the last line?

pisicaverde, 2017-05-13:
Indeed there is a huge RAM consumption... Reverted back to u8glib...

olikraus, 2017-05-14:
Sure. 
But future are the standard I2C and SPI libs. I can not write my own ROM optimized I2C and SPI procedures for each new controller and board.

olikraus, 2017-05-14:
ToDo: Implement u8g.getStrWidthP()

SukkoPera, 2017-05-14:
Can we make a short analysis of how much RAM is used by which library? Because 350 bytes seem a lot to me. The Wire and SPI libraries on the Uno platform look quite optimised.

olikraus, 2017-05-14:
For which microcontroller?

SukkoPera, 2017-05-14:
ATmega328 at least. That's where I get 350 bytes extra, sorry I forgot to mention.

olikraus, 2017-05-14:
I have attached the map file from this project:
https://github.com/olikraus/u8g2/tree/master/sys/arm/stm32l031x6/u8g2_rtc

[u8g2_rtc.txt](https://github.com/olikraus/u8g2/files/999338/u8g2_rtc.txt)

It has an ARM controller STM32L0 (Cortex-M0+). The controller is connected to a 128x64 SSD1306 (Full buffer setup).

In this map file you find two sections, one is for data, the other for bss segment:
```
Allocating common symbols
Common symbol       size              file
key_queue           0x10              u8g2_rtc.a(key.o)
rtc                 0x12              u8g2_rtc.a(main.o)
u8g2                0x80              u8g2_rtc.a(main.o)
```

```
 *(.bss*)
 .bss.Tamper2Count
                0x20000004        0x4 u8g2_rtc.a(main.o)
                0x20000004                Tamper2Count
 .bss.select_pos
                0x20000008        0x4 u8g2_rtc.a(main.o)
                0x20000008                select_pos
 .bss.RTCIRQCount
                0x2000000c        0x4 u8g2_rtc.a(main.o)
                0x2000000c                RTCIRQCount
 .bss.SysTickCount
                0x20000010        0x4 u8g2_rtc.a(main.o)
                0x20000010                SysTickCount
 .bss.Tamper3Count
                0x20000014        0x4 u8g2_rtc.a(main.o)
                0x20000014                Tamper3Count
 .bss.cursor_pos
                0x20000018        0x4 u8g2_rtc.a(main.o)
                0x20000018                cursor_pos
 .bss.buf.4808  0x2000001c        0x4 u8g2_rtc.a(u8x8_u8toa.o)
 .bss.key_queue_end
                0x20000020        0x1 u8g2_rtc.a(key.o)
                0x20000020                key_queue_end
 .bss.key_queue_start
                0x20000021        0x1 u8g2_rtc.a(key.o)
                0x20000021                key_queue_start
 .bss.buf.7435  0x20000022      0x400 u8g2_rtc.a(u8g2_d_memory.o)
```

In this project u8g2 contributes with exactly two objects: u8g2  (0x80=128 Bytes, u8g2 main structure) and .bss.buf.7435 (0x400= 1024 bytes, display buffer for the SSD1306).

In the AVR world, a little bit different. Read-only data is not automatically placed into flash memory. Instead on AVR uC this has to be done manually. From ARM map file:

```
 .rodata.u8x8_u8toa_tab
                0x0800253b        0x3 u8g2_rtc.a(u8x8_u8toa.o)
 *fill*         0x0800253e        0x2 
 .rodata.u8g2_cb_r0
                0x08002540        0x8 u8g2_rtc.a(u8g2_setup.o)
                0x08002540                u8g2_cb_r0
 .rodata.u8x8_d_ssd1306_128x64_noname_powersave0_seq
                0x08002548        0x5 u8g2_rtc.a(u8x8_d_ssd1306_128x64_noname.o)
 .rodata.u8x8_d_ssd1306_128x64_noname_powersave1_seq
                0x0800254d        0x5 u8g2_rtc.a(u8x8_d_ssd1306_128x64_noname.o)
 *fill*         0x08002552        0x2 
 .rodata.u8x8_ssd1306_128x64_noname_display_info
                0x08002554       0x18 u8g2_rtc.a(u8x8_d_ssd1306_128x64_noname.o)
 .rodata.u8x8_d_ssd1306_128x64_noname_flip0_seq
                0x0800256c        0x7 u8g2_rtc.a(u8x8_d_ssd1306_128x64_noname.o)
 .rodata.u8x8_d_ssd1306_128x64_noname_init_seq
                0x08002573       0x35 u8g2_rtc.a(u8x8_d_ssd1306_128x64_noname.o)
 .rodata.u8x8_d_ssd1306_128x64_noname_flip1_seq
                0x080025a8        0x7 u8g2_rtc.a(u8x8_d_ssd1306_128x64_noname.o)
 *(.eh_frame*)
                0x080025b0                . = ALIGN (0x4)
```
This is 80025b0−800253b = 0x075 = 117 Bytes, which would be required additionally on the AVR architecture at the moment.

Summery (ignoring the memory for the display buffer):

128 Bytes       Core data for (display information&state, font decoding data, ...)
10 Bytes         Add on for the Arduino U8g2 Class (Cursor postion, Arduino.print support)
117 Bytes       Display command sequences and display information structure. Placed in RAM for AVR, but will end up in Flash memory for all other architectures.

olikraus, 2017-05-20:
hmmm... there is also no DrtawStrP function.
Is this really required. Probably yes, but only AVR architectures have this problem.

SukkoPera, 2017-05-22:
Uhm, actually [I got around that using setCursor() and print()](https://github.com/SukkoPera/MyTrackr/commit/a160ea4094dde1befe84afc6135a34d0929cdbe7#diff-0d87297b49711ce049a85c181ba89494L85).

olikraus, 2017-05-22:
Yes, makes sense, when using the F Makro.

olikraus, 2017-05-22:
ok, closing this for now, let me wait what other users think

closed 2017-05-22 by olikraus