olikraus/u8g2 #263: Can't compile with "U8X8_SSD1306_128X64_NONAME_HW_I2C(U8X8_PIN_NONE);"

camp0s, 2017-05-10:
Hello, this happes and I can't see where I'm wrong :(

The display worked with:
U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(SCL, SDA, U8X8_PIN_NONE);

I wanted to switch to HW I2C with this:
U8X8_SSD1306_128X64_NONAME_HW_I2C(U8X8_PIN_NONE);

..\Documents\Arduino\libraries\U8g2\src/clib/u8x8.h:309:23: error: expected ')' before numeric constant
 #define U8X8_PIN_NONE 255
                                          ^
Did I compose the constructor the wrong way?

olikraus, 2017-05-11:
Can you provide more information? Ide version? Board type? uC?

gnbl, 2017-05-14:
Regarding that constructor (NONAME or VCOMH0) resulting in error: while

```
#include <U8x8lib.h>
U8X8_SSD1306_128X64_NONAME_HW_I2C u8g2(U8X8_PIN_NONE);
```
 works for me with 2.14.7 on 1.8.2 for Nano clone (m328p),

omitting the [supposedly optional argument](https://github.com/olikraus/u8g2/wiki/u8x8setupcpp/f73965511ab30ee3a0d8407ed7fb82f5ef3ffbc0#ssd1306-128x64_noname-1) like `U8X8_SSD1306_128X64_VCOMH0_HW_I2C u8g2();` results in

```
exit status 1
request for member 'begin' in 'u8g2', which is of non-class type 'U8X8_SSD1306_128X64_VCOMH0_HW_I2C()'
```
although the code seems to provide a default argument:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.h#L367

<hr/>

BTW, the link in https://github.com/olikraus/u8g2/wiki/u8x8setupcpp#constructor-name (no source view / line links of Wiki pages on GitHub?) is broken, it should point to https://github.com/olikraus/u8g2/wiki/u8x8setupcpp#constructor-reference  and also, that list is unsorted :-)

olikraus, 2017-05-14:
regarding the links: both links 
https://github.com/olikraus/u8g2/wiki/u8x8setupcpp#constructor-name
and
https://github.com/olikraus/u8g2/wiki/u8x8setupcpp#constructor-reference
exist. Not sure what is broken here.

gnbl, 2017-05-14:
The link _in_ the section I linked first 

> All available construcor names are listed in the reference section below.

currently links https://github.com/olikraus/u8g2/wiki/u8x8setupcpp#reference which _does not  exist_.

olikraus, 2017-05-14:
I get the same error. However i have no idea about what is wrong here. I am also not a C++ expert..
```
HelloWorld.cpp: In function 'void setup()':
HelloWorld.cpp:141:8: error: request for member 'begin' in 'u8x8', which is of non-class type 'U8X8_SSD1306_128X64_NONAME_HW_I2C()'
   u8x8.begin();
        ^
```
However, you have workaround, right? Just provide the default value as first argument.

olikraus, 2017-05-14:
Links: Ah, thanks, fixed.

olikraus, 2017-05-20:
no further input from OP... closing

closed 2017-05-20 by olikraus

camp0s, 2017-05-21:
Sorry for the absence, taking another look at my constructor:

U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(SCL, SDA, U8X8_PIN_NONE);
U8X8_SSD1306_128X64_NONAME_HW_I2C(U8X8_PIN_NONE);

..I forgot the "u8x8" before the parameters field -.-

olikraus, 2017-05-21:
Ok, thanks for the feedback.