olikraus/u8g2 #264: Only half of display written, other half shows noise (solved)

gnbl, 2017-05-13:
Sorry to open a non-issue, but I think the information might be interesting to others and I could't find info on a forum/group to post to.

Display: 
"1pcs 0.96 inch 128X64 OLED Display Module For arduino 0.96 IIC I2C Communicate white"
https://www.aliexpress.com/item/1pcs-0-96-white-0-96-inch-OLED-module-New-128X64-OLED-LCD-LED-Display-Module/32640136921.html

with four pins.

I was using the Adafruit Library (https://github.com/adafruit/Adafruit_SSD1306 & https://github.com/adafruit/Adafruit-GFX-Library) with limited success, as the display seemed to show only every other line (interlaced) making me wonder whether the line distance on cheap OLEDs might really be that bad.

But then I tried u8g2 (Wiki "Hello World!" sample code using `U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2( U8G2_R0 );`) to find that the display is capable of showing neighbouring vertical pixels albeit with some flickering in the top and with the bottom half of the display showing noise ("snow") but not updating.

The u8g2 library was first installed via manager in 1.8.2 and later overwritten with files from master (insert clib path for includes). Using the sample code from wiki intro with modified constructor (see above).

It wasn't a firmware issue: after power-cycling the display everything works fine. So the previous code (or rather my settings) apparently upset it :-) It does not have a reset line, so fingers crossed..

gnbl, 2017-05-13:
Let's hope this gets still crawled/indexed so people can find it.

closed 2017-05-13 by gnbl

gnbl, 2017-05-13:
VCOMH0 (`U8G2_SSD1306_128X64_VCOMH0_1_HW_I2C`) gives much more noticable dim values (looping from 0..127. 

```
uint8_t c = 0;
...
u8g2.setContrast(c);
if(++c>127)c=0;
```

gnbl, 2017-05-14:
My bad with Adafruit library - it's source needs to be configured. For this display, the default size needs to be changed by un/commenting these #defines:

https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.h#L73

This resolves the every-2nd-line issue described above.

I first looked into TWI addresses, but everything is alright:
The I2C address to use with the Adafruit library (based on Wire) is 0x3C (decimal 60) which gets left-shifted somewhere (I think in Wire) to make space for the R/W bit on the bus, so this works for me:

`display.begin(SSD1306_SWITCHCAPVCC, 0x3c);`

while u8g2 uses the left-aligned value of 0x78 or 120 decimal, as revealed by

`    uint8_t getI2CAddress() { return u8g2_GetI2CAddress(&u8g2); }`

added to https://github.com/olikraus/u8g2/blob/master/cppsrc/U8g2lib.h#L67 and then output as 

```
  u8g2.begin();
  Serial.begin(9600);
  while (!Serial);
  Serial.print( u8g2.getI2CAddress() );
```

(since I could not find it in the code).

olikraus, 2017-05-14:
The fact that u8g2 used the left-aligned version of the i2c address is described also in the reference manual for u8g2. 

Indeed, compared to the Adafruit lib, i tried to develop u8g2 in such a way, that the user does not need to modify the lib sources.