olikraus/u8g2 #265: Concurrency enable feature

PiotrC1, 2017-05-16:
Hi,
Firts of all thanks for a great library. I'm using Arduino Nano with ATmega 328. I'm communicating with analog to digital converter in interrupt form timer T1 via I2C bus every 10 ms. In my main program I'm using 128x64 SH1106 OLED display. Communication with I2C in interrupt breaks communication with display in main loop. So a kind of semaphore saying "I2C busy now" put on/off both via main program and timer interrupt is necssary. Timer interrupts are every 10 ms, so blocking I2c via U8G2 has to be done for as short period as possible.

Best regards,
Piotr

olikraus, 2017-05-16:
What does "break communication" mean? SW I2C or HW I2C?

PiotrC1, 2017-05-17:
To break communication it means that there are artefact appearing on the screen (skipped lines, distorted characters etc.). It happens only while timer I2C (btw. operated by Wire library) reads / writes are made.
I'm using hardware I2C communication (both for SH1106 display and ADC).
Timer interrupts happen every 10ms, ADC reads every 70ms. My idea is to "inform" I2C reads/writes in main loop (with a kind of semaphore) 10ms before interrupt will access I2C to finish and wait for semaphore to be off. Just an idea.

olikraus, 2017-05-17:
The point is: If you read your sensor values within the picture loop, then you might see similar effects like distorted chars. It is very important, that you read the sensor value before the picture loop into a variable, so that the sensor value is stable troughout the complete loop. This effect has nothing todo with your interrupts. I think i have mentioned this in the FAQ also.

olikraus, 2017-05-21:
Conclusion:
For HW I2C, Wire lib should take care on interrupt issues. OP did not clearly mention the original problem, so it might be caused by a wrong usage of the picture loop also.

Feel free to reopen this...

closed 2017-05-21 by olikraus