olikraus/u8g2 #266: adding more language fonts

jasoneppink, 2017-05-17:
Hi @olikraus - I really appreciate all of your work on this amazing library.

I'm using u8g2 for a museum [subtitling project](https://github.com/jasoneppink/extsub). The Arabic, Chinese, and Cyrillic support is invaluable, but I'd also love to include Bengali (Unicode U+0980 - U+09FF), Korean (Unicode U+AC00 - U+D7AF), and Urdu (Unicode U+11080 - U+110CF).

Can you provide some insight/documentation on how fonts are created and prepared for u8g2? I'd love for these to be supported, and to help make that a reality if I can.

olikraus, 2017-05-17:
I assume the fonts should be based on unifont, right?
I can add the mentioned font support to u8g2 lib. You can create fonts by yourself with the tool bdfconv.exe. See FAQ for this.

jasoneppink, 2017-05-17:
Ah! Thanks for mentioning the FAQ. (Duh.)

Yes, if you could add support for these languages using unifont that would be amazing. The full blocks for each of these languages is below. I know you needed to cull glyphs from the Chinese for memory reasons. I suspect a similar decision might need to be made for the Korean/Hangul glyphs, but I am too ignorant of the language to assist with that right now.

Bengali:
U+0980 to U+09FF

Korean (Hangul):
U+AC00-U+D7A3
U+1100–U+11FF
U+3130-U+318F
U+A960-U+A97F
U+D7B0-U+D7FF

Urdu:
U+0600 to U+06FF
U+0750 to U+077F
U+FB50 to U+FDFF
U+FE70 to U+FEFF

olikraus, 2017-05-20:
hmm...the fontsize for korean would be more than 350K flashrom.
I think i can only support a subset here.

olikraus, 2017-05-21:
Maybe someone can do frequency analysis.
Korean content: http://www.hani.co.kr/

We should also include all glyphs from here:
https://www.topikguide.com/download/6000_korean_words.htm

Two learning korean textbooks are mentioned here:
https://www.reddit.com/r/languagelearning/comments/1yv0nd/free_korean_textbooks_over_900_pages_pdf_and/
We could analyse the hangul glyphs in these text books also and include them.

olikraus, 2017-05-21:
added bengali and urdu. I think for hangul we need more research: Goal should be to have a list of the most common hangul chars (maybe top 1000, top 2000)

olikraus, 2017-05-21:
Hangul/Korean still missing, moving this to future release

xyz37, 2017-06-28:
Hi @olikraus(I'm very grateful for your efforts)  and @jasoneppink.
I also want to express Hangul as a Korean.
I read your dialogs and read FAQ BDFConv article.
And some test.
I created a BDF file with a public font (2,258,177bytes), and I ran it on Windows with the following command, but the amount of generated c file (6,240bytes) is very small.

> bdfconv -f 1 "32" -n NanumGothicCoding-Bold-R-Normal--16 -o NanumGothicCoding-Bold-R-Normal--16.c ..\OTF2BDF\Output\나눔고딕코딩-Bold_16.bdf

Another question is what to enter the value of the -m option.

olikraus, 2017-06-28:
Probably, the "32" is wrong. I assume it will be interpreted as imput file.

Maybe this works better:
```
bdfconv -f 1 -n NanumGothicCoding-Bold-R-Normal--16 -o NanumGothicCoding-Bold-R-Normal--16.c ..\OTF2BDF\Output\NanumGothicCoding-Bold-R-Normal--16.bdf
```

However, indeed you need the -m command.
-m specifies one or multiple ranges within the unicode space. "-m" defines which unicodes are taken from the .bdf file and copied to the u8g2 font.

For example -m "32,48-57" will select space (unicode 32) and the digits "0" to "9" (unicode 48 to 57).
If you want unicode range U+AC00-U+D7A3 to be included, use -m "32,48-57,$ac00-$d7a3" (you can prefix numbers with $ for hex notation)

It is also possible to put the commands or individual glyph into an external file and read the data from there. In this case, use "-M <filename>"

xyz37, 2017-06-30:
Thank you.
At first I did not understand your answer.
I've also removed the -m value and -m "32, 48-57, $ ac00- $ d7a3".
Both methods are not applicable.
However, English is an displayed, but Hangul does not appear.
The Unicode range of the font used is $ AC00- $ D7A3 correct.
You can refer to the chinese.map file.
I extracted the all Unicode value that actually had a value from the BDF file that converted the Korean TTF font.
I converted to korean.map file and it succeeded.
However, the converted 16x16 size font file is about 1.1mb too large.

I need to resize the finished file.
If possible, I would like to contribute to your U8g2 library.

Would you like to test it?
I made github repository [https://github.com/xyz37/U8G2_Korean_Font](https://github.com/xyz37/U8G2_Korean_Font)

olikraus, 2017-06-30:
I think the following command should work:

bdfconv -v -f 1 -m "32-128,44032-55215" unifont.bdf -o u8g2_korean.c  -n u8g2_korean

I have used decimal numbers instead of hex numbers, which probably works better.
Can you provide the koran.map file?

olikraus, 2017-06-30:
I found two lists with a reduced number of glyphs, which might be usefull:
https://pastebin.com/aSzg7Twa  about 320 glyphs
https://pastebin.com/GMgaqZkR about 2000 glyphs

olikraus, 2017-07-01:
With the map files above, i have create new fonts and added them to u8g2. Fonts are base on unifont bdf file.

https://github.com/olikraus/u8g2/wiki/fntgrpunifont

xyz37, 2017-07-03:
I am grateful.
We've tested the beta version, but not all characters.
That character is included in korean2.map.
For example, '녕' is a value of $B155 and is not output on the screen.
'녕' is a word like hello, but '녕' is not output.
I'll check out some more.

olikraus, 2017-07-03:
Yes, this would be great. Just report back with a list of codes, which seem to be important from your side. Then I will add these codes to korean1.map and korean2.map

yksht, 2017-07-12:
Hi. I converted cyrillic font "HaxrCorp 4089 Cyrillic AltGr Regular" to work with u8g2. I used it in my home project but didn't found any info about its license. I would be glad to share converted font with comunity. Could I commit it? And are there problems with license?

olikraus, 2017-07-13:
Yes, I need to know from where the original sources came from, who is the author and what is the license for the font.

yksht, 2017-07-13:
I got this font from here: https://fonts2u.com/haxrcorp-4089-cyrillic-altgr-regular.font 
Here are links to "fontshop.com" but all links are broken ((
Designer's nickname is sahwar. Also didn't found any info about designer.

olikraus, 2017-07-13:
Which options did you use to convert the font with bdfconv?

yksht, 2017-07-13:
bdfconv -f 1 -m "32-255,1025-1105" -n haxrcorp_4089_cyr_altgr_reg -o haxrcorp_4089_cyr_altgr_reg.c haxrcorp_4089_cyr_altgr_reg.bdf

olikraus, 2017-07-13:
Ok, I think it can be added to the next release

olikraus, 2017-07-15:
added $B155 to korean1, $B155 already was included in korean2

olikraus, 2017-08-11:
korean is now implemented

closed 2017-08-11 by olikraus