olikraus/u8g2 #267: Support for SSD1327 96 x96 Gray Oled Display

OBorro, 2017-05-17:
Hi All,

I am using two seeed Gray Oled display and Arduino Mega - Grove Shield - I2C port to show different messages in each display. Those display only have one address 0x3C.  Using I2C port, the pins for clock and data are 21 and 20. I already checked in other forum about it, but I didn't find nothing about it. So any idea to make that happen? This is the code I have:

//U8G2_SSD1327_SEEED_96X96_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);  // Seeedstudio Grove OLED 96x96
//U8G2_SSD1327_SEEED_96X96_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);  // Seeedstudio Grove OLED 96x96

U8G2_SSD1327_SEEED_96X96_F_SW_I2C d1(U8G2_R0, 21, 20);
U8G2_SSD1327_SEEED_96X96_F_SW_I2C d2(U8G2_R0, 21, 20);

void setup(void) {
  
  d1.begin();{
  d1.clearBuffer();          // clear the internal memory
  d1.setFont(u8g2_font_ncenB10_tr); // choose a suitable font
  d1.drawStr(0,20,"Voltage #1");  // write something to the internal memory
  d1.sendBuffer();          // transfer internal memory to the display
  delay(1000);
  }
  {

  d2.begin();
  d2.clearBuffer();          // clear the internal memory
  d2.setFont(u8g2_font_ncenB10_tr); // choose a suitable font
  d2.drawStr(0,20,"Voltage #2");  // write something to the internal memory
  d2.sendBuffer();          // transfer internal memory to the display
  delay(1000);
}
}
void loop(void) {
  }

olikraus, 2017-05-17:
If the address is the same, then the displays must not share the same gpio pins. You need four pins in this cases.

OBorro, 2017-05-17:
Thanks for your help. The arduino grove mega shield has 3 port I2C and I using just two of them. Is there a way I can change the I2C address to separate the displays? Could you provide me more details about the previous comment? I do not have to much experience working with arduino.

olikraus, 2017-05-17:
With U8g2 SW_I2C you can use ANY pins of the Arduino Mega. You are not restricted to any special ports. But i guess this would require to cut the wires of your displays and connect them directly to some ports.

olikraus, 2017-05-17:
Another option could be to use
U8G2_SSD1327_SEEED_96X96_F_HW_I2C d1(...)
and
U8G2_SSD1327_SEEED_96X96_F_2ND_HW_I2C d2(...)

olikraus, 2017-05-17:
excause me... some i always hit the close button

OBorro, 2017-05-17:
Don't worry. Ok, I will try that and then I let you know what happen. Thank you for your support.

olikraus, 2017-05-17:
you mean this shield?
https://www.generationrobots.com/img/cms/GorveMegaShield/800px-Megashield001.jpg

olikraus, 2017-05-17:
you could also do this:

U8G2_SSD1327_SEEED_96X96_F_SW_I2C d1(U8G2_R0, 10, 11);
U8G2_SSD1327_SEEED_96X96_F_SW_I2C d2(U8G2_R0, 8, 9);

and connect displays into J11 and J12

OBorro, 2017-05-17:
Yes, That is correct. Ok. I will try that.

OBorro, 2017-05-17:
Thank you so much for your help. It is working finally with the last option that you gave me. Problem solved.

olikraus, 2017-05-17:
Great

closed 2017-05-17 by olikraus