olikraus/u8g2 #268: Working with a larger image than can be displayed

Tulsa2008, 2017-05-17:
What I am trying to do is to load a relatively large image in the computer's memory and display small sections of the image at a time.  

I will clarify:  
Assuming that I am using a 128 by 64 pixel display.  
I want to load an image that is 400 by 400 pixels in RAM (Just for example). 
I would like to be able to display, display sized (128 x64 pixels) sections of the image and move around in the image using the “X” “Y” coordinates arguments.  

I think the buffer used by the display would have to be made much larger but maybe not.  

I am using an Arduino Mega with an external memory board.  This allows me to have SRAM available in 56k byte chunks.

My hope is to be able to load maps in memory from and SD chip and be able to move around within the map as needed.

Is this possible??  
Ken

olikraus, 2017-05-18:
You could use setPixel to copy pixel values from your large picture to the display

Tulsa2008, 2017-05-18:
Thanks for the input.  In order for me to access the larger image the way I want, I am converting from a single dimensional array to a two dimensional array.  Your comment got me thinking and I may be able to send a row at a time.  

Again, thanks for the input.

Ken

olikraus, 2017-05-18:
Finally you have to go down to pixel level. I do not see any other way here. 
One optimization could be to combine multiple pixel of the same color into a drawHLine command. drawHLine is probably the fastest way to communicate with u8g2.

Tulsa2008, 2017-05-18:
Again, Thanks for your input.  I will give both a try.  

I really do like your lib.  You spend a lot of time on it and a LOT of people appreciate it.

Ken

closed 2017-05-19 by Tulsa2008