olikraus/u8g2 #274: U8G2_ST7920_128X64_1_6800 not working

Croutonix, 2017-05-22:
I get an error with the constructor `U8G2_ST7920_128X64_1_6800(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset])`

```
test:22: error: expected ')' before ',' token

 U8G2_ST7920_128X64_1_6800(U8G2_R2, 2, 3, 4, 5, 6, 7, 8, 9, 13, 12, 11);
                                  ^

test:22: error: expected unqualified-id before numeric constant

 U8G2_ST7920_128X64_1_6800(U8G2_R2, 2, 3, 4, 5, 6, 7, 8, 9, 13, 12, 11);
                                    ^
```

Both U8g2 and U8x8 are included. The serial constructor works but not this one in parallel.

Croutonix, 2017-05-22:
Never I just forgot `u8g2` before

closed 2017-05-22 by Croutonix

olikraus, 2017-05-22:
yes.. indeed...