olikraus/u8g2 #275: Incompatibility between u8g2 and ucglib

viti95, 2017-05-25:
I'm trying to use two screens in one single proyect, a SH1106 and a ILI9163 at the same time, but both are incompatible (Arduino 1.8.2, U8G2 2.15.2, Ucglib 1.4.0). Sample code imports both U8g2lib.h and Ucglib.h, this is the error:

> 
> libraries\Ucglib\clib\ucg_font.c.o (symbol from plugin): In function `ucg_font_calc_vref_font':
> (.text+0x0): multiple definition of `u8g_GetFontBBXOffX'
> libraries\U8g2\clib\u8g2_font.c.o (symbol from plugin):(.text+0x0): first defined here
> c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions
> libraries\Ucglib\clib\ucg_polygon.c.o (symbol from plugin): In function `pg_ClearPolygonXY':
> (.text+0x0): multiple definition of `pg_ClearPolygonXY'
> libraries\U8g2\clib\u8g2_polygon.c.o (symbol from plugin):(.text+0x0): first defined here
> libraries\Ucglib\clib\ucg_polygon.c.o (symbol from plugin): In function `pg_ClearPolygonXY':
> (.text+0x0): multiple definition of `pg_AddPolygonXY'

olikraus, 2017-05-25:
Good point. Yes, I think this is a bug.

olikraus, 2017-07-01:
fixed

closed 2017-07-01 by olikraus

i-make-robots, 2018-07-19:
Hello from 2018.  I found this page because I have similar errors, conflicting between u8glib and u8g2.  I'm only using u8g2 in this project but have the other for Marlin support.  I updated all my arduino libraries and the problem persists.